bdring/Grbl_Esp32 #597: Multiple spindel modes in parallel (Laser PWM & Servo PWM)

eokgnah, 2020-09-15:
**Please describe the feature you would like implemented**
i would like to be able to configure servo pwm and spindle pwm on different pins.
so if i send for example M03 S500 i would get on one pin a laser with 50% power and on the other pin a servo moving to 90 degrees.

if i have attached the laser, it will burn
if i have attached the servo it will lift the pen
same gcode, different heads.
 
**Why do you think this would improve Grbl_ESP32?**
even more useful to use different plug and play hardware without needing a different gcode or other machine settings for an other head.

**What do you need the feature for?**
i have a simple X/Y only cnc with interchangeable laser (engraving&cutting) and servo (pen lifting) heads.
the servo and the laser are connected to different pins.
did not find how to set up my my_machine.h for this double parallel config.
 
**Will this feature appear to a lot of users?**
maybe, think so. ;-)

bdring, 2020-09-18:
We are working on a feature to dynamically change hardware features. This means you could probably create a macro to change modes at the click of a button in the WebUI.

We have no plans to have multiple spindle types active at the same time.

dnmnsmith, 2020-10-13:
Looking forward to it.

I'm doing the same, switching between  Laser and Relay type spindles on different pins. It's tricky in the code as it stands, as they are both PWM variants, so they pick up the same pin definitions.

My solution at the moment is a default constructor on PWM passing the pins:

       PWM(uint8_t output_pin = UNDEFINED_PIN, uint8_t enable_pin = UNDEFINED_PIN);

The default in PWM is to take the normal pin definitions (SPINDLE_OUTPUT_PIN etc) if the pins are UNDEFINED, as above or accept these values if set.

Laser can then over-ride the pins for its usage:

        Laser() : PWM( LASER_OUTPUT_PIN, LASER_ENABLE_PIN ) { }

However, I suspect the best solution is more general purpose than this; with the more general purpose config settings the pins would probably better be set that way for each spindle type. I don't want to offer my changes for merge for this reason, but if anyone wants to use this in the meantime until the official solution is avaible, do shout.