bdring/Grbl_Esp32 #599: controlling eleksmaker a3 laser

lamikr, 2020-09-16:
I just assembled eleksmaker a3 laser and would like to change it's mana controll board which runs grpl on arduino to grbl_esp32  (grbl upload over wifi, faster mcu, ...) 
I tried to look up the documentation and boards available currently on Tindie but before purchase I would like to verify couple of things where I could not find clear answer from wiki.

1) To my understanding I need support only for 2 stepper motors (x and y with either tmc2208/tmc2209/tmc2160) and the laser with the eleksmaker a3, so any grbl_esp32 board available on Tindie with support for 2 or more stepper motors would be enough?

2)  have one 300 mW laser module which takes power on eleksmaker board from gnd and 12V pins Can that be connected somehow directly to grbl_esp32 board? 

I have just also ordered 500 mW pwm controlled laser module (gnd, 12v and pwm pins connected to eleksmaker mana board). Does the grbl_esp32 support this type of lasers on hw and software and where it would be connected?

3) Do I need to do lot of stepmotor or laser specific changes to firmware(pwm or no pwm, step motor current, etc...) to source code? I am used to code/build/compile/flash with arduino-ide but I am not familiar with the grbl or grbl_esp32 source code.

4) These are the boards I find to be now available on Tindie. Would any of them work ok with eleksmaker laser with pwm control?
a)  Pen Plotter Controller for ESP32 & TMC2130 V2
https://www.tindie.com/products/33366583/pen-plotter-controller-for-esp32-tmc2130-v2/
b)  Grbl_ESP32 CNC Development Board V4.1
https://www.tindie.com/products/33366583/grbl_esp32-cnc-development-board-v41/
c) 6 Pack Universal CNC Controller
Newest one, maybe a little overkill for the laser but could be good upgrade to my sovol sv01 3d printer?

MitchBradley, 2020-09-16:
I have the 4.1 board connected to an Eleks-laser.  It's helpful to have three stepper drivers so you can use a separate driver for each of the side motors.  That gives better performance than using a single driver for two motors, like on the EleksMaker Mana boards. Grbl_Esp32 supports axis ganging so that is easy to configure. 

My laser has a TTL PWM pin that is separate from the power pins, so I was able to directly drive that from the 4.1's PWM signal, with an external 2.2K resistor to ground to keep the laser turned off when the system is booting.

If/since your laser has only 12V and GND, you would need some extra circuitry to drive it with PWM.  You could use a logic-gate  MOSFET like IRLR120N or IRLR6225, with Source connected to GND, Gate to the PWM pin on the board, and Drain connected to the negative terminal on the 2-pin laser connector, with the positive laser terminal connected to +12V.  I would add a pulldown resistor between Gate and Source to ensure that it is off during startup.  Also a switch on the 12V line so you can quickly turn it off.

lamikr, 2020-09-16:
Thanks for the info, Which pins you have used for connecting the lasers gnd, power and pwm pins to your 4.1 development board?

I noticed that there is also available a separately sold modules like "0-10V Spindle CNC I/O Module"
  https://github.com/bdring/Grbl_Esp32/issues/599
I assume those can somehow be connected to development board as well?

MitchBradley, 2020-09-17:
For PWM I used the pin labeled SpinPWM and for GND I used the pin adjacent to it, labeled GND - both on the 10-pin connector.  For 12V you can use the VMot pin on the 4-pin connector - but I soldered a wire directly to the power jack.

The various Modules only work with the 6-pack controller board.  The 6-pack board is intended to be a universal board that can serve a wide variety of CNC machines via the Modules, but for now it is would not be suitable for a 3D printer because there are no modules for the special 3D printer needs like temperature control.  I have a long-term plan to make such a module, but there is no schedule yet.  There would also be some software work to do because 3D printing is mostly done with Marlin.  Grbl_Esp32 has many features for milling and other CNC modalities, but does not cater to the special needs of 3D printing like temperature control, extruder and filament management, bed leveling, ....

closed 2020-09-17 by MitchBradley