bdring/Grbl_Esp32 #600: Soft limits and servos

gautamjain, 2020-09-16:
**Firmware Version:**
1.3a Date 20200828

**Is the problem repeatable?**
Yes

**Under what conditions does the bug occur?**
I'm using the ESP32 Dev Controller v4.1 for testing.  I'm also using a servo for Z instead of a stepper:

```C++
#define Z_SERVO_PIN 		    GPIO_NUM_2
#define Z_SERVO_RANGE_MIN	    0.0
#define Z_SERVO_RANGE_MAX	    360.0

// Temporarily enabled positive machine space because I don't have limit switches, but I still 
// want to use soft limits.  I "manually" home before powering on the ESP32 (i.e. by moving the
// gantry by hand to machine zero).
#define HOMING_FORCE_POSITIVE_SPACE
```

When soft limits are enabled, I cannot travel farther than Z100.  It seems like [`system_check_travel_limits()` in `System.cpp` ](https://github.com/bdring/Grbl_Esp32/blob/981ce50993cfc0a59ad83d7a9c4bff8457db5a51/Grbl_Esp32/src/System.cpp#L242-L245) is checking to see if the move would be outside `$Z/MaxTravel`.  And by default, `$Z/MaxTravel` is 100.

I would have expected soft limits to be checked against `Z_SERVO_RANGE_MIN` and `Z_SERVO_RANGE_MAX` for servos.

**Boot messages**
```
[MSG:Grbl_ESP32 Ver 1.3a Date 20200828]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:DEV_GANTRY_V0.2]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(21) Disable:None Index:-1]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(26) Dir:GPIO(15) CS:GPIO(25) Disable:None Index:-1]
[MSG:Z  Axis RC Servo motor Output:2 Min:0.000mm Max:360.000mm]
[MSG:Global stepper disable pin:GPIO(13)]
[MSG:X  Trinamic driver test passed]
[MSG:Y  Trinamic driver test passed]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:No spindle]
[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
Grbl 1.3a ['$' for help]
```

bdring, 2020-09-16:
Soft limits uses the MaxTravel. In the next release of Grbl_ESP32 you will set the range using of the servo using MaxTravel.

There is a major overhaul to the MaxTravel, Positive Space and Force Zero type options. That is [detailed here](https://github.com/bdring/Grbl_Esp32/issues/575).

I am curious why you wanted the Z servo travel to be different than the MaxTravel.

gautamjain, 2020-09-16:
Thanks, I'll read through those changes.

Maybe I misunderstood how servos should be setup.  I thought `MaxTravel` and `StepsPerMm` are percent values (used for calibrating the servo's endpoint positions).  

My servo has an operating range of a little over 360 degrees.  I want to map 1 mm of Z travel to 1 degree of servo rotation.   That's why I set `Z_SERVO_RANGE_MIN` to 0.0 and `Z_SERVO_RANGE_MAX` to 360.0.

After that I was going to adjust `StepsPerMm` and `MaxTravel` so the servo endpoints limited the actual movement to exactly 360 degrees.  They might end up being something like 105% and 95%.

bdring, 2020-09-16:
There are separate calibration methods in the new method. I used those
other settings because Grbl settings are difficult to change. They had
horrible names for calibration. We are overhauling the settings to allow
many more settings and setting types.

Initially the calibration will be hardcoded in the machine definition and
soon nothing will be hard coded, not even the pin number 😱

m3chanist, 2020-09-29:
"and soon nothing will be hard coded, not even the pin number"

Hurrah!

closed 2020-11-02 by bdring