bdring/Grbl_Esp32 #603: Feature request: Probe Monitoring toggle via gcode.

jschoch, 2020-09-17:
**Please describe the feature you would like implemented**

GCode command to turn probe pin monitoring on and off

**Why do you think this would improve Grbl_ESP32?**

When doing multi point probing via a sender/macro the probe may contact the work when moving between probe start positions.  The following edge finding example:

```
G91F400
G1F500X9
G1F500Z-6
G38.3F400X-5
G0X1
G38.3F200X-5
G1F500X1
G53G1F500Z-3.826
G53G1F500X1.2
G1F500Y-9
G1F500Z-6
G38.3F400Y5
G0Y-1
G38.3F200Y5
G1F500Y-1
G53G1F500Z-3.826
G53G1F500Y6.431
```

Each non-G38 move has the potential to crash the probe.  

Another example is heightmap scanning where many moves are made to scan a PCB or other surface.

**What do you need the feature for?**

the longevity of my probe

**Will this feature appear to a lot of users?**

anyone who is creating probe macros for multi point probing.

bdring, 2020-09-17:
Please explain the problem behavior and exactly what you want to happen.

What does prove on/off do for you?

I would rather not guess

jschoch, 2020-09-17:
here is a reference from linuxcnc  https://github.com/LinuxCNC/linuxcnc/issues/368

The behavior I am looking for is that on the start of a probing macro grbl will turn probe pin monitoring on for non-G38 movements.  This will stop the machine if it attempts to move the probe through the work while moving probing start positions.

From the above example.  I am probing and edge.  

```
G1F500X9  ;  Move from Start position to "clear" part
G1F500Z-6 ;  Move down 6 
G38.3F400X-5 ; probe
```

The probe may crash on line 1 or line 2 and grbl would just keep going even though the probe was triggered on the pre-G38 positioning move.  If probe pin monitoring was active on these intermediate moves grbl could halt and go into alarm when the probe makes contact.

jschoch, 2020-09-17:
another option here is to have a probe detect pin which would detect the probe and ensure that the probe pin state is monitored when the probe is "on".  This could also be used to lockout turning the spindle on.

MitchBradley, 2020-09-17:
There is some similarity to the safety door code.

bdring, 2020-09-17:
Was that LinuxCNC feature implemented?

MitchBradley, 2020-09-17:
A solution that requires no GCode involvement is to have the probe pin act as a limit switch when not probing.

jschoch, 2020-09-17:
> Was that LinuxCNC feature implemented?

```
// not probing, but we have a rising edge on the probe.
// this could be expensive if we don't stop.
```

https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/motion/control.c#L620

Since recording the position per step ISR tick is expensive in the default mode for probe state monitoring, perhaps it is better to have a new routine that only checks a probe state flag in the same interval used for limit switch checking when not in an G38.  I believe the expense of keeping the current stat monitor running is why it is not always on.  I would also expect that monitoring probing full time combined with  bad wiring and EMI noise will create newbie user issues.

jschoch, 2021-04-04:
here's why I wanted to have this available.  I was probing a corner.   The macro was unaware the probe triggered, there is no way to stop it if the machine is moving fairly quickly, and so the machine just kept moving Z down.

[image]

bdring, 2021-04-04:
If the probe were a reset switch most of the time and a probe switch when you needed it to be, that might fix your problem.

With the current code, that would rather messy. Reset switches use interrupts and probes do not. If you wired 2 inputs to the switch that would help, but you would need to turn off the reset during probing. Custom macro programming might do it.

I don't think it is something the developers would want to implement at this time. We have a lot of other things we want to work on.

freisei, 2021-04-20:
maybe of interest for you:
https://forum.linuxcnc.org/24-hal-components/42321-new-component-probefilter-to-handle-probe-tripped-errors#206504