bdring/Grbl_Esp32 #615: Endstop

topogigio, 2020-09-26:
Hi,
thanks fo your great job.
I'm trying to test endstop, I double check pin assignment and hw change status un esp32 pin.
I don't thing that's an issue, but for me doesn't work, what I miss...
How can I configure separate enable motor for each axis?

My machine conf file at the end.
Regads
Andrea

#define MACHINE_NAME            "4axis_MotoMic"

#ifdef N_AXIS
        #undef N_AXIS
#endif
#define N_AXIS 4

#define X_STEP_PIN             GPIO_NUM_18		//Height
#define X_DIRECTION_PIN   GPIO_NUM_17		//Height
#define Y_STEP_PIN              GPIO_NUM_26		//Width
#define Y_DIRECTION_PIN    GPIO_NUM_25		//Width
#define Z_STEP_PIN              GPIO_NUM_12		//Depth
#define Z_DIRECTION_PIN    GPIO_NUM_14		//Depth
#define A_STEP_PIN              GPIO_NUM_23	//Rotation
#define A_DIRECTION_PIN    GPIO_NUM_22	//Rotation

#define STEPPERS_DISABLE_PIN    GPIO_NUM_19		//Test only Height motor

#define SPINDLE_TYPE            SpindleType::PWM // only one spindle at a time

#define X_LIMIT_PIN             GPIO_NUM_39		//Height
#define Y_LIMIT_PIN             GPIO_NUM_34		//Width
#define Z_LIMIT_PIN             GPIO_NUM_35		//Depth
#define A_LIMIT_PIN             GPIO_NUM_36		//Rotation

bdring, 2020-09-29:
You can map up to 2 switch input per axis like...

```C++
#define X_LIMIT_PIN      GPIO_NUM_17
#define X2_LIMIT_PIN     GPIO_NUM_2
```

They are linked together in firmware. You will only see the X axis switch triggered in the status regardless of which switch was triggered. It is the same as wiring 2 switches to the same input assuming you wire them correctly for N.O. and N.C modes.

See this wiki page for more info

https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-Limit-Homing-Switch#machine-definition

topogigio, 2020-09-29:
Thanks for prompt reply, Pn:X and other axes are showed when I pressed each endstop. 
Btw, notting happen when motor spin and I press endstop.

bdring, 2020-09-29:
Endstops will only stop motion during homing and when used as hard limits.

closed 2020-09-30 by bdring