bdring/Grbl_Esp32 #623: TMC5161: could use some help

roberthuitema, 2020-10-02:
Hi Guys,

I am trying to get my 4-Axis SPI CNC controller board V1.3 board to function with a set of TMC5161 drivers I bought from Bigtreetech and some Nema23 motors (2.80A/phase) for my CNC project. 

Everything works fine, except the motors stop moving when I set the amps to around 0.4A or higher in the settings (the standard setting of 0.25A usually works to some extend). When I give a jog command the motors do some jittering/vibrating but are not rotating. Sometimes this happens immediately, sometimes it happens after a few movements. All four drivers display this behavior.  Also sometimes when this happens a driver keeps drawing current and gets hot. After a reset and I guess reconfiguration of the driver they do function again (until it goes wrong again). 

My setup:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20200928]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:SPI_DAISY_4X_xyyz]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 Limits(0.000,0.000)]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 Limits(0.000,0.000)]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:3 Limits(0.000,0.000)]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:4 Limits(0.000,0.000)]
[MSG:X  Trinamic driver test passed]
[MSG:Y  Trinamic driver test passed]
[MSG:Y2 Trinamic driver test passed]
[MSG:Z  Trinamic driver test passed]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:PWM spindle Output:GPIO(25), Enbl:GPIO(4), Dir:None, Freq:5000Hz, Res:13bits]
[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:X  Axis limit switch on pin GPIO(36)]
[MSG:Y  Axis limit switch on pin GPIO(39)]
[MSG:Z  Axis limit switch on pin GPIO(34)]
Grbl 1.3a ['$' for help]
```
and GRBL settings:

```
$0 = 3    (Step pulse time, microseconds)
$1 = 250    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 200.000    (Homing locate feed rate, mm/min)
$25 = 2000.000    (Homing search seek rate, mm/min)
$26 = 250.000    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1.000    (Maximum spindle speed, RPM)
$31 = 0.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$N1=
$N0=
$33 = 5000.000   
$34 = 0.000   
$35 = 0.000   
$36 = 100.000   
$100 = 1066.666    (X-axis travel resolution, step/mm)
$101 = 1066.666    (Y-axis travel resolution, step/mm)
$102 = 1066.666    (Z-axis travel resolution, step/mm)
$103 = 100.000   
$104 = 100.000   
$105 = 100.000   
$110 = 3000.000    (X-axis maximum rate, mm/min)
$111 = 3000.000    (Y-axis maximum rate, mm/min)
$112 = 3000.000    (Z-axis maximum rate, mm/min)
$113 = 1000.000   
$114 = 1000.000   
$115 = 1000.000   
$120 = 50.000    (X-axis acceleration, mm/sec^2)
$121 = 50.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$123 = 200.000   
$124 = 200.000   
$125 = 200.000   
$130 = 300.000    (X-axis maximum travel, millimeters)
$131 = 300.000    (Y-axis maximum travel, millimeters)
$132 = 300.000    (Z-axis maximum travel, millimeters)
$133 = 300.000   
$134 = 300.000   
$135 = 300.000   
$140 = 0.250   
$141 = 0.250   
$142 = 0.200   
$143 = 0.250   
$144 = 0.250   
$145 = 0.250   
$150 = 0.125   
$151 = 0.125   
$152 = 0.125   
$153 = 0.125   
$154 = 0.125   
$155 = 0.125   
$160 = 8   
$161 = 8   
$162 = 8   
$163 = 16   
$164 = 16   
$165 = 16   
$170 = 16   
$171 = 16   
$172 = 16   
$173 = 16   
$174 = 16   
$175 = 16 
```

What I have tried so far:

- Use definition of the TMC5160's in spi_daisy_4axis_xyyz.h and changed the value of TMC5160_RSENSEDEAULT to 0.062f (as the sense registers are 0.062 Ohm)
-Copied some code to be able to define the drivers as 5161's and not get a message that they are unsupported drivers.
```
 if (_driver_part_number == 2130) {
            tmcstepper = new TMC2130Stepper(cs_pin, _r_sense, spi_index);
        } else if (_driver_part_number == 5160) {
            tmcstepper = new TMC5160Stepper(cs_pin, _r_sense, spi_index);
	} else if (_driver_part_number == 5161) {
            tmcstepper = new TMC5161Stepper(cs_pin, _r_sense, spi_index);
        } else {
            grbl_msg_sendf(CLIENT_SERIAL, MsgLevel::Info, "%s Axis Unsupported Trinamic part number TMC%d", _axis_name, _driver_part_number);
            has_errors = true;  // as opposed to NullMotors, this is a real motor
            return;
        }
```

- use a variable bench power supply at 24V and 10A and a standard 24V 9A Meanwell power supply.
- Using less microstepping (8)
- Playing around with the current settings ($140-$142)
- Using StealthChop mode instead of CoolStep

This all without much success and I am out of ideas how to get the drives to function properly. So I was hoping one of you would have some ideas or suggestions I can try next. :) Any help would be much appreciated!

bdring, 2020-10-02:
Several people have reported problems with the TMC5160 at higher currents and voltages. At this time we believe it has something to due with the register values in the firmware.

https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Motors/TrinamicDriver.cpp#L207

We are not experts on that chip or the settings and have not been able to find a solution. Trinamic sells an evaluation kit and some software that helps tune the drivers. It is $200 and not in the budget yet.

We do not believe it has anything to do with the hardware at this time.

roberthuitema, 2020-10-02:
Thanks for your reply. So I understand correctly you have been able tested with the 5160's at lower currents only as well?

bdring, 2020-10-02:
I personally have been able to use them all the way up to 3.5A @12V & 24V. Other people cannot get above about 0.6A.

My drivers are BigTreeTech brand and use 0.075 Ohm resistors. People with trouble are also using BigTreeTech.

roberthuitema, 2020-10-02:
Ok I just tried my second set of TMC5161 drivers (I accidentally received my order twice), but unfortunately they behave exactly the same. Thanks for your help, if I manage to find a way to get them working I will post it here.

KingMo5h, 2020-10-02:
I had a similar problem with my original TMC5160. But my problem was that the power supply wasnt set right. (amp to low). I'm not sure anymore but i also had some trouble with microstepping, set Microsteps (SPI Drivers ) to 255 or 256 dont know anymore. After that the steppers worked fine.

Also set Motor current (SPI Drivers ) to 2.80 in defaults.h and Motor hold current (SPI Drivers ) to something low. Don't know if this got fixed/reworked.

@bdring 
Wouldn't it be better if the defaults.h is also in the machines list? So you have every setting in one data? For example all defaults.h settings in i2s_out_xyzabc_trinamic.h or 3axis_xyz.h. And in there only the options that are relevant for the "Board" or "machine"? 
Would be awsome if you dont have to search the common settings.

bdring, 2020-10-03:
@KingMo5h The items in Defaults.h are only used if you have not defined them in your machine definition file.

[midtbot.h](https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Machines/midtbot.h#L56) is a good example of one with a lot of setting defaults.

roberthuitema, 2020-10-03:
I tried some more microstep settings and changing the values in defaults.h, unfortunately this doesn't work for me. 

However when I change the number of micro steps it actually seems to change the steps/mm instead? At least my motors spin faster or slower when changing the micro step value (higher microstep value = slower/less rotation), while I did not change any other setting. What am I doing wrong here?

lav8org, 2021-02-28:
When I had a problem with 2130s/5160s crashing (stepping stops but no apparent error registered on the driver as far as serial debug revealed; power cycle reqd to revive stepping) StealthChop was the culprit; disabling it entirely and just using SpreadCycle ended the problem. I had found a Trinamic rep posting comments somewhere (just tried earnestly to find it again, but no dice) saying that it was caused by stops/reversals too big/sudden for StealthChop tripping a safety. IIRC they claimed StealthChop2 (as found on the 5160) would be better at handling the situation, but on that count my mileage definitely varied.

I've experienced the crash on two setups: BTT SKR 1.3 w/BTT TMC2130 at 24V 600+ mA, and BTT SKR 1.4 w/BTT TMC5160 12V logic and 34V 600+ mA motor. I've run both a fair bit without any driver crashes... except the few occasions I succumbed to temptation and re-enabled StealthChop.