bdring/Grbl_Esp32 #627: Homing Issues

Mboss32, 2020-10-06:
Hi All, 

I am running into errors while trying to home using CNCjs.  The main HOME button that issues a '$H' command triggers an "error:18".  The single axis homing buttons (G28.2) commands trigger an "error:20 (unsupported command)".  Similarly the $H in WebUI will cause an "error:18" but the single axis variants like $HX will perform properly.  

I see the Homing cycle settings have been pulled out of the Config.h file, but I can seem to find them in settings.h or settings.cpp, any suggestions?   

What version of the firmware are you using?  Using the latest commit from Oct 6, 2020

Is the problem repeatable? Yes

bDuthieDev, 2020-10-06:
$Homing/Cycle5=
$Homing/Cycle4=
$Homing/Cycle3=
$Homing/Cycle2=Y
$Homing/Cycle1=X
$Homing/Cycle0=Z

Mboss32, 2020-10-06:
Thanks @bDuthieDev!  What file did you add the to?

bDuthieDev, 2020-10-06:
Just enter from a console. Use $S to see all your current settings.

bdring, 2020-10-06:
We are slowly moving as much as we can to run time settings to reduce the need to compile. We support all the traditional $$ menu numeric settings and a newer text base settings style. You can see all the new settings by sending $S.

It is a lot of settings, but if you can filter it. Send $cycle to see any setting with the word cycle in it. Just be careful to avoid existing commands like $X.

Also, to find descriptions of error codes, send $E to see all of them or $E=18 to see a specific number.

bdring, 2020-10-06:
You can home multiple axes at once. A traditional homing might be.

$Homing/Cycle0=Z
$Homing/Cycle1=XY

More on settings is here...

https://github.com/bdring/Grbl_Esp32/wiki/Settings

Mboss32, 2020-10-06:
Thanks!  Is there a way to add these to defaults.h the settings so they reload after RST commands?

bdring, 2020-10-06:
Yes,

They are bit masks. The most readable way to do it is like this.

#define DEFAULT_HOMING_CYCLE_0      bit(Z_AXIS)
#define DEFAULT_HOMING_CYCLE_0      (bit(X_AXIS) | bit(Y_AXIS))

Mboss32, 2020-10-06:
Thanks for the suggestion Bart!  

Unfortunately, I have tried the above in my machine file, and in defaults.h and it does not work.  The defaults do not load after compiling/upload or RST commands.  

I also tried this method from the 6_pack_stepstick_XYZ_v1.h example (pleased in machine file).   

#ifdef HOMING_CYCLE_0
    #undef HOMING_CYCLE_0
#endif
#define HOMING_CYCLE_0 bit(Z_AXIS) // Z first

#ifdef HOMING_CYCLE_1
    #undef HOMING_CYCLE_1
#endif
#define HOMING_CYCLE_1 (bit(X_AXIS)|bit(Y_AXIS))

None of these methods appear to work, the settings are still blank after each test.

bdring, 2020-10-06:
Could you paste your complete machine definition file here. I will give a test on a controller.

Mboss32, 2020-10-06:
Sure thing - it is the SPI_daisychain with just a few mods. Thanks!

#define MACHINE_NAME "SPI_DAISY_4X_xyyz"

#ifdef N_AXIS
        #undef N_AXIS
#endif
#define N_AXIS 3 // can be 3 or 4. (if 3 install bypass jumper next to the A driver)

#define TRINAMIC_DAISY_CHAIN

#define TRINAMIC_RUN_MODE           TrinamicMode :: CoolStep
#define TRINAMIC_HOMING_MODE        TrinamicMode :: CoolStep

#ifdef HOMING_CYCLE_0
    #undef HOMING_CYCLE_0
#endif
#define HOMING_CYCLE_0 bit(Z_AXIS) // Z first

#ifdef HOMING_CYCLE_1
    #undef HOMING_CYCLE_1
#endif
#define HOMING_CYCLE_1 (bit(X_AXIS)|bit(Y_AXIS))

// Use SPI enable instead of the enable pin
// The hardware enable pin is tied to ground
#define USE_TRINAMIC_ENABLE

#define DEFAULT_HOMING_SQUARED_AXES 0

// Y motor connects to the 1st driver
#define X_TRINAMIC_DRIVER       2130        // Which Driver Type?
#define X_RSENSE                TMC2130_RSENSE_DEFAULT
#define X_STEP_PIN              GPIO_NUM_12
#define X_DIRECTION_PIN         GPIO_NUM_14
#define X_CS_PIN                GPIO_NUM_17  // Daisy Chain, all share same CS pin

// Y motor connects to the 2nd driver
#define Y_TRINAMIC_DRIVER       2130        // Which Driver Type?
#define Y_RSENSE                TMC2130_RSENSE_DEFAULT
#define Y_STEP_PIN              GPIO_NUM_27
#define Y_DIRECTION_PIN         GPIO_NUM_26
#define Y_CS_PIN                X_CS_PIN  // Daisy Chain, all share same CS pin

// Y2 motor connects to the 2nd driver
#define Y2_TRINAMIC_DRIVER       2130        // Which Driver Type?
#define Y2_RSENSE                TMC2130_RSENSE_DEFAULT
#define Y2_STEP_PIN              GPIO_NUM_15  // Z on schem
#define Y2_DIRECTION_PIN         GPIO_NUM_2   // Z on schem
#define Y2_CS_PIN                X_CS_PIN  // Daisy Chain, all share same CS pin

// Z Axis motor connects to the 4th driver
#define Z_TRINAMIC_DRIVER       2130        // Which Driver Type?
#define Z_RSENSE                TMC2130_RSENSE_DEFAULT
#define Z_STEP_PIN              GPIO_NUM_33 // A on schem
#define Z_DIRECTION_PIN         GPIO_NUM_32 // A on schem
#define Z_CS_PIN                X_CS_PIN  // Daisy Chain, all share same CS pin

// Mist is a 3.3V output
// Turn on with M7 and off with M9
#define COOLANT_MIST_PIN        GPIO_NUM_21

#define SPINDLE_TYPE            SpindleType::PWM
#define SPINDLE_OUTPUT_PIN         GPIO_NUM_25
#define SPINDLE_ENABLE_PIN      GPIO_NUM_4

// Relay operation
// Install Jumper near relay
// For spindle Use max RPM of 1
// For PWM remove jumper and set MAX RPM to something higher ($30 setting)
// Interlock jumper along top edge needs to be installed for both versions
#define DEFAULT_SPINDLE_RPM_MAX     12000 // Should be 1 for relay operation

#define PROBE_PIN               GPIO_NUM_22

#define X_LIMIT_PIN             GPIO_NUM_36
#define Y_LIMIT_PIN             GPIO_NUM_39
#define Z_LIMIT_PIN             GPIO_NUM_34

bdring, 2020-10-06:
Not at my computer right now, but change this

```
#ifdef HOMING_CYCLE_0
    #undef HOMING_CYCLE_0
#endif
#define HOMING_CYCLE_0 bit(Z_AXIS) // Z first

#ifdef HOMING_CYCLE_1
    #undef HOMING_CYCLE_1
#endif
#define HOMING_CYCLE_1 (bit(X_AXIS)|bit(Y_AXIS))
```

to this

``` 
#define DEFAULT_HOMING_CYCLE_0 bit(Z_AXIS)
#define DEFAULT_HOMING_CYCLE_1 (bit(X_AXIS) | bit(Y_AXIS))
```

Mboss32, 2020-10-06:
No joy using the suggested edits

#define DEFAULT_HOMING_CYCLE_0 bit(Z_AXIS)
#define DEFAULT_HOMING_CYCLE_1 (bit(X_AXIS) | bit(Y_AXIS))

Its like the $Homing/Cycle setting variables aren't being assigned from the #defines, I have been looking for this in the code but haven't found the right section yet.

MitchBradley, 2020-10-06:
```
diff --git a/Grbl_Esp32/src/SettingsDefinitions.cpp b/Grbl_Esp32/src/SettingsDefinitions.cpp
index 2f649dc..eb3ad77 100644
--- a/Grbl_Esp32/src/SettingsDefinitions.cpp
+++ b/Grbl_Esp32/src/SettingsDefinitions.cpp
@@ -377,7 +377,9 @@ void make_settings() {
 
     const char* homing_names[] = { "Homing/Cycle0", "Homing/Cycle1", "Homing/Cycle2",
                                    "Homing/Cycle3", "Homing/Cycle4", "Homing/Cycle5" };
-    for (uint8_t i = 0; i < MAX_N_AXIS; i++) {        
-        homing_cycle[i] = new AxisMaskSetting(EXTENDED, WG, NULL, homing_names[i], 0);
+    const int32_t homing_defaults[] = { DEFAULT_HOMING_CYCLE0, DEFAULT_HOMING_CYCLE1, DEFAULT_HOMING_CYCLE2,
+                                        DEFAULT_HOMING_CYCLE3, DEFAULT_HOMING_CYCLE4, DEFAULT_HOMING_CYCLE5 };
+    for (uint8_t i = 0; i < MAX_N_AXIS; i++) {
+        homing_cycle[i] = new AxisMaskSetting(EXTENDED, WG, homing_defaults[i], homing_names[i], 0);
     }
 }
```

bdring, 2020-10-06:
I am sorry, that feature is brand new this week and still in the devt branch. I am hoping to merge that with the main branch soon, but you could use it now to fix your problem.

bdring, 2020-10-07:
Merged with main....closing

closed 2020-10-07 by bdring

Mboss32, 2020-10-07:
Thanks all for the help!  I will test the new main out in the morning