bdring/Grbl_Esp32 #631: Abort during jogging

ExploratoryDevices, 2020-10-06:
What version of the firmware are you using?
Grbl_ESP32 Ver 1.3a Date 20200823

Is the problem repeatable?
Intermittent, but has happened several times. It is preventing me from using this firmware reliably as the reboot loses all my work offsets.

Under what conditions does the bug occur?
Jogging using bCNC. Runs fine for minutes at a time, crashes appear random. Sometimes will run for 10-15 minutes without a reboot. Serial terminal reports two reboots. Backtrace and decoding below. Modified config.h with disabled wifi and bluetooth.

Raw serial report. Previous command was a G1 from bCNC.
```
abort() was called at PC 0x40144923 on core 1
Backtrace: 0x4008dca0:0x3ffb1e40 0x4008ded1:0x3ffb1e60 0x40144923:0x3ffb1e80 0x4014496a:0x3ffb1ea0 0x4014475d:0x3ffb1ec0 0x401444e0:0x3ffb1ee0 0x400d5789:0x3ffb1f00 0x400d5d84:0x3ffb1f20 0x400d645d:0x3ffb1f40 0x400d2c47:0x3ffb1f70 0x400d190b:0x3ffb1f90 0x400dfdd9:0x3ffb1fb0 0x4008a30d:0x3ffb1fd0
Rebooting...
[MSG:Grbl_ESP32 Ver 1.3a Date 20200823]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:OMIO 4-Axis Router]
[MSG:Axis count 4]
[MSG:RMT Steps]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d3ccd  PS      : 0x00060031  A0      : 0x800811d9  A1      : 0x3ffbe790
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffc1258
A6      : 0x3ffb0000  A7      : 0xbaad5678  A8      : 0x3ffc1258  A9      : 0x00000018
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x8008d262  A13     : 0x3ffb1e00
A14     : 0x00000001  A15     : 0x3ffb51c4  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb
Core 1 was running in ISR context:
EPC1    : 0x400d3ccd  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008747d
Backtrace: 0x400d3ccd:0x3ffbe790 0x400811d6:0x3ffbe7b0 0x40084df9:0x3ffbe7d0 0x4000bfed:0x3ffb1e70 0x4008b479:0x3ffb1e80 0x400e3d27:0x3ffb1ea0 0x4013c55e:0x3ffb1ee0 0x40081145:0x3ffb1f10 0x400db0e3:0x3ffb1f50 0x400d2b76:0x3ffb1f70 0x400d1903:0x3ffb1f90 0x400dfdc7:0x3ffb1fb0 0x4008a30d:0x3ffb1fd0
Rebooting...
[MSG:Grbl_ESP32 Ver 1.3a Date 20200823]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:OMIO 4-Axis Router]
[MSG:Axis count 4]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:X  Axis standard stepper motor Step:GPIO(4) Dir:GPIO(2) Disable:None]
[MSG:Y  Axis standard stepper motor Step:GPIO(26) Dir:GPIO(15) Disable:None]
[MSG:Z  Axis standard stepper motor Step:GPIO(27) Dir:GPIO(33) Disable:None]
[MSG:A  Axis standard stepper motor Step:GPIO(18) Dir:GPIO(14) Disable:None]
[MSG:B  Axis standard stepper motor Step:GPIO(16) Dir:GPIO(17) Disable:None]
[MSG:Global stepper disable pin:GPIO(13)]
[MSG:DAC spindle Output:GPIO(25), Enbl:GPIO(22), Dir:None, Res:8bits]
[MSG:Probe on pin GPIO(32) Inverted:Y]
Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]
```

```
PC: 0x400d3ccd: motors_set_direction_pins(unsigned char) at C:\Users\imamo\AppData\Local\Temp\arduino_build_660561\sketch\src\Motors\Motors.cpp line 370
EXCVADDR: 0x00000000
```

Decoded stack results (first reboot)
```
0x4008dca0: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008ded1: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x40144923: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x4014496a: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x4014475d: __cxxabiv1::__cxa_allocate_exception(std::size_t) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc line 268
0x401444e0: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x400d5789: system_execute_line(char*, unsigned char, WebUI::AuthenticationLevel) at C:\Users\imamo\AppData\Local\Temp\arduino_build_660561\sketch\src\ProcessSettings.cpp line 606
0x400d5d84: execute_line(char*, unsigned char, WebUI::AuthenticationLevel) at C:\Users\imamo\AppData\Local\Temp\arduino_build_660561\sketch\src\Protocol.cpp line 82
0x400d645d: protocol_main_loop() at C:\Users\imamo\AppData\Local\Temp\arduino_build_660561\sketch\src\Protocol.cpp line 169
0x400d2c47: run_once() at C:\Users\imamo\AppData\Local\Temp\arduino_build_660561\sketch\src\Grbl.cpp line 119
0x400d190b: loop() at C:\Users\imamo\Dropbox\Stuff\Projects\CNC\Grbl_Esp32-main\Grbl_Esp32/Grbl_Esp32.ino line 28
0x400dfdd9: loopTask(void*) at C:\Users\imamo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x4008a30d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

Decoded stack results (second reboot)
```
0x400d3ccd: motors_set_direction_pins(unsigned char) at C:\Users\imamo\AppData\Local\Temp\arduino_build_660561\sketch\src\Motors\Motors.cpp line 370
0x400811d6: onStepperDriverTimer(void*) at C:\Users\imamo\AppData\Local\Temp\arduino_build_660561\sketch\src\Stepper.cpp line 256
0x4008b479: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x400e3d27: esp_intr_alloc_intrstatus at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/intr_alloc.c line 666
0x4013c55e: timer_isr_register at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/timer.c line 206
0x40081145: Stepper_Timer_Init() at C:\Users\imamo\AppData\Local\Temp\arduino_build_660561\sketch\src\Stepper.cpp line 1222
0x400db0e3: stepper_init() at C:\Users\imamo\AppData\Local\Temp\arduino_build_660561\sketch\src\Stepper.cpp line 450
0x400d2b76: grbl_init() at C:\Users\imamo\AppData\Local\Temp\arduino_build_660561\sketch\src\Grbl.cpp line 41
0x400d1903: setup() at C:\Users\imamo\Dropbox\Stuff\Projects\CNC\Grbl_Esp32-main\Grbl_Esp32/Grbl_Esp32.ino line 24
0x400dfdc7: loopTask(void*) at C:\Users\imamo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 14
0x4008a30d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

bdring, 2020-10-07:
Can you try with the latest version of the main branch? That is what I will be looking at.

ExploratoryDevices, 2020-10-07:
Updating firmware now and will attempt and update if this problem re-occurs.

Note that the enums for "spindleTypes" are not backwards compatible with the version I was using. I needed to update my machine file from "SPINDLE_TYPE_DAC" to "SpindleType::DAC". This hasn't been updated in the documentation yet.

bdring, 2020-10-07:
I updated the wiki.

If you can repeat the problem on the latest main branch, please try to find an easy way to reproduce it.  Give the actual commands sent to Grbl_ESP32, not the bCNC actions. I already have about 10 senders installed. Not excited about installing another one.

ExploratoryDevices, 2020-10-07:
I flashed the latest version last night and ran several jobs through it with no crashes over the course of about 6 hours. I will update here if it happens again with the log of commands if it happens again.

bdring, 2020-10-07:
My testing went fine too.

You mentioned you were losing your offsets when it crashed. That should not happen. Can you explain what was lost? Are you using G92?

ExploratoryDevices, 2020-10-08:
I may have misspoken. It's the machine position that's lost during a reboot, requiring re-homing. I'm not confident enough yet to know whether re-homing would allow me to resume the job without re-setting the work position so I treated it as "lost". Sorry for the incorrect terminology.

closed 2020-11-02 by bdring