bdring/Grbl_Esp32 #634: PCB or Engraving Milling Probing

leadgtr7, 2020-10-09:
**Please describe the feature you would like implemented**
I would like the ability to probe a surface and modify the loaded g code to account for surface inconsistencies
**Why do you think this would improve Grbl_ESP32?**
It would allow rapid PCB prototyping with small trace width and spacing. This would allow users to prototype new hardware quicker and provide easier revisioning
**What do you need the feature for?**
I would like it for PCB Milling and Engraving
**Will this feature appear to a lot of users?**
I would think anyone who wanted to make PCBs and Engraving would really appreciate it

In addition if it helps here are some software programs that have implemented this

OpenCNCpilot
bCNC
chilliPeppr
universal gCode Sender

While I can use the above programs to do what i need It would be much easier to have this ability built into the web gui so that I can run these types of jobs without being tethered to a computer.

leadgtr7, 2020-10-09:
I had imagine this being implemented more like an autobed level routine on a 3d printer. My thought would be the user could pick a certain number of points (say 5x10 or whatever really). Then a table of x and y with z offsets could be used to bi linear interpolate.

I didn't think something like that would take much resource as most 3d printers running even 8 bit marlin can do this

MitchBradley, 2020-10-09:
Even so, it is probably better to do it in WebUI instead of in the controller

leadgtr7, 2020-10-09:
That was where I would like it to be. Not sure if this was the right place for it though

MitchBradley, 2020-10-09:
The WebUI code is in https://github.com/luc-github/ESP3D-WEBUI , maintained by a different developer

leadgtr7, 2020-10-10:
I have put an issue with the ESP3D-WEBUI github page

closed 2020-10-10 by leadgtr7