bdring/Grbl_Esp32 #640: Solenoid setup problems

cbvicious, 2020-10-12:
Attempting to get an ESP32 Pen/Laser Controller up and running with a solenoid. It appears there may have been some changes along the way. 
In pen_laser.h, for example:
```
For pen mode be sure to uncomment #define USE_PEN_SERVO in config.h
--
For solenoid  mode be sure to uncomment #define USE_PEN_SERVO in config.h
```
This option no longer exists in config.h. Also, there is a solenoid.h file, but there is nothing at all that calls for it to be used, and no documentation that I can find.

Any assistance getting this up and running would be appreciated. I really hope this is me missing something simple.

bdring, 2020-10-12:
The solenoid pen feature became obsolete when we moved to motor classes in the firmware. You can control a solenoid with several other methods. 

- Using a relay spindle
- With mist and flood coolant using M7,M8, & M9
- As a digital output using M62 and M63
- As a PWM output with M67

A solenoid motor class could be added, but I don't think any developers would see it as a priority at this time.

cbvicious, 2020-10-12:
So, the board which I bought from you on tindie, a few months ago, is no longer usable?

cbvicious, 2020-10-12:
Would a reasonable solution be to run the MotorClass branch from the GIT?

bdring, 2020-10-12:
Do you have a machine designed yet?  I think a servo makes a better axis than a solenoid.

If you need the solenoid axis, I could add it. Do you just need the solenoid to activate when the axis is above 0? Is you solenoid rated for continuous duty?

cbvicious, 2020-10-12:
I bought the Pen/Laser controller board, specifically to restore an old pen
plotter machine which uses a solenoid. Conversion to a servo would require
removal of the safety shield. I considered that option, but I have children
around the house and don't want little fingers to enter moving parts.

The solenoid should be rated for constant duty, as far as i know.

The solenoid should only need to be active during pen/blade down
conditions.

>

bdring, 2020-10-12:
Can you send a snippet of the g-code that goes from pen up to pen down and back?

cbvicious, 2020-10-12:
I am hoping to use this tool.

https://www.calvertcreativesolutions.com/category-s/146.htm

> Oddities to "Normal" CNC:
> X Axis - rollers to pull vinyl in and out
> Y Axis - carriage
> M3 - Solenoid On
> M5 - Solenoid Off
> M8 - Vacuum Fans On
> M9 - Vacuum Fans Off
> S - PWM for solenoid pressure
> F - Feedrate instead of rapid move * Optional
> All other commands should work as expected.

I have no interest in the fans commands

bdring, 2020-10-12:
We already support that. Use this in your machine definition.
```
#define SPINDLE_TYPE               SpindleType::PWM
#define SPINDLE_OUTPUT_PIN         GPIO_NUM_16
#define DEFAULT_SPINDLE_RPM_MAX    1000.0   // this should be the max of the S parameter range you are using.
```

cbvicious, 2020-10-12:
That appears to be working, but is there any way to implement the holding power function which was available with solenoid control?

bdring, 2020-10-12:
We would not want to do that in firmware. M3/M5 is spindle g-code and your request does not make sense for spindle.

It could be done in g-code
- **S100** (full power)
- **M3** (turn on)
- **G4P0.25** (wait 1/4 second)
- **S50** (reduce power)