bdring/Grbl_Esp32 #641: X axis coordinate is 10.000mm bigger than actually position

bd7jat, 2020-10-12:
I use 20201004 main branch. 
This problem is repeatable.
It happans under any condition.
After system starts up ,input "?" and  get a response coordinate as "X0 Y0 Z0",that's fine.
But if i input any X axis value ,the response X positon is 10.00mm bigger than the order i gave,
let's say , i input "G00X10Y0Z0", gives me "ok",then input"?" to check the coordinate, esp32 returns me "X20Y0Z0"
Even  i just input "G00x0Y0Z0",  esp32 returns me "X10Y0Z0"
How can i figure this out ? Did i do something wrong when compiling?
Thanks for you guys great job.

bd7jat, 2020-10-12:
It's casued by WCO ,my fault

closed 2020-10-12 by bd7jat

bd7jat, 2020-10-12:
I can not find a way to changer WCO of X axis ?Any suggestion ?Thanks

bdring, 2020-10-12:
The G10 L20 <http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l20>
g-code is the easiest way to change your WCO. Most senders including the
WebUI have buttons for it.