bdring/Grbl_Esp32 #642: WebUI: unexpected data in Commands window

bDuthieDev, 2020-10-12:
When going to the GRBL tab and then pressing the update button, the results of a $$ command are shown in the Commands window in the Dashboard tab. Other options in other tabs do not cause unexpected results to show up in the Commands window. Behaves the same in both main and Devt branches

luc-github, 2020-10-12:
Every grbl commands output (not [ESPXXX]) go to terminal because it is when they go to terminal that they are handled 

It is not a bug - it is a feature ^_^

In other hand if you do `$$` in terminal - it will also automaticaly update the GRBL tab content

MitchBradley, 2020-10-12:
Based on luc's answer, this is not a Grbl_Esp32 issue, so closing.

closed 2020-10-12 by MitchBradley