bdring/Grbl_Esp32 #643: WebUI and Grbl_ESP32

InvncibiltyCloak, 2020-10-14:
Hey, 
I am looking into adding support for the Maslow CNC on top of this project. 
Step zero is to try building the test drive, which worked fine when calling `./build-machine.py test_drive.h`.
However, this is not a complete build since it does not include the WebUI part of the project. This left me with a few questions:

1. Is the contents of the `embedded/` folder supposed to be the 'compiled' webpage of WebUI used with Grbl_ESP32?
    - If so, I assume the file should be renamed `index.html.gz` and uploaded. However this did not work for me.
2. Can we use an updated, latest release of WebUI?
    - If so, should the `embedded/` folder be removed?
    - Should the `src/WebUI` folder be a git submodule linking to the @luc-github repo?
3. Any thoughts about setting it up such that `index.html.gz` gets built and uploaded automatically?

I can submit a PR if any of this needs to get cleaned up.

luc-github, 2020-10-14:
1 - no this is the embedded page in FW itself - it is a maintenance page in to update filesystem content or Fw itself and it is embedded in code itself, index.html.gz is for the main UI 
2 - the repository project of webUI is https://github.com/luc-github/ESP3D-webui but you can use any that fit your needs
 having a maintenance page instead of full ui is done because simple is more robuste for maintenance than complex App it also allows to use any UI and seperate FW from UI - but anything is possible it is just matter of choice even I do not recommend.
src/webUI could be a submodule but this was not a need because the compiled file is already in data directory - so you can upload directly when you flash also

Additionnaly current GRBL_ESP32 has a flash size limitation when enabling Bluetooth and WiFi (around 144KB for flash and so for webUI files)  so you cannot build all current supported languages but only a subset, I am working on v3.0 that will support language packs, instead of embedded all languages in index.html.gz, which should give more flexibility
 
3 - as mentionend upload can be done when flashing, it upload the content of data directory tool exists for platformIO and Arduino IDE, this will depend of partitions size. 

That said this is on my feedback- may be @bdring has others plans

luc-github, 2020-10-14:
I also enclose coming features plan included in ESP3D 3.0 and ESP3D-WEBUI 3.0: https://github.com/luc-github/ESP3D/blob/3.0/ESP3D-features.xls
Feel free to do feedback as GRBL part is not yet started - if anything is missing or needed please raise it

MitchBradley, 2020-10-14:
You might consider storing that file in .csv formation instead of .xls, so it can be viewed as text without downloading and going through the Microsoft dance.

luc-github, 2020-10-14:
it is done by libreoffice , csv cannot handle colors

InvncibiltyCloak, 2020-10-15:
Okay thanks for the help. 
Thanks for pointing out that the index.html.gz is hidden away in the `Grbl_Esp32/src/data` folder. I might make a PR to update some of the build instructions in README. 

Also, I didnt realize space was so tight that only a single language fits. The "upload after build" procedure makes a lot more sense in that context.

I might try to automate uploading of this file if I get a chance, but for now I will close out this issue. Thanks again everyone!

closed 2020-10-15 by InvncibiltyCloak

luc-github, 2020-10-15:
Please note that GRBL_ESP32  is not single language but 9 :
en,  fr, es,  de , it, pl, ptbr, ru, uk

currently total available 13:
en,  fr,  es,  de,  it, ja, hu, pl, ptbr, ru, tr, uk,  zh_cn,

But if you flash without BT, or use a board with more flash, all languages should fit