bdring/Grbl_Esp32 #645: Soft limits enable/disable per axis

ExploratoryDevices, 2020-10-18:
**Please describe the feature you would like implemented**
If it doesn't already exist (I couldn't find any mention of a mask in the wiki), it would be handy to be able to enable/disable soft limits on a per-axis basis.

I have a 4th (A) axis for my desktop router which is running GRBL-ESP32. I'm trying to figure out the best way to set up this axis in terms of limits/homing. I've set the MaxTravel to the highest value (100000). The XYZ axes have limit switches. Soft limits are enabled. Unfortunately, this means that my A axis is only able to rotate in one direction at start because it boots up at position "0" and soft limits prevent it from moving past zero in the negative direction. If the axis were homed, I could just set the /Home/Mpos to the middle of the range and probably not run into the limits in a normal job, but there's no limit switch on it so I have no way of triggering the setting of Mpos. (Alternatively, is there a way to set the default "cold start" Mpos of an axis?)

**Why do you think this would improve Grbl_ESP32?**
GRBL-ESP32 adds support for more than three axes, which in many applications will introduce a combination of linear and rotary axes (with and without limit switches). Soft limits are no longer global.

**What do you need the feature for?**
Enable use of continuous axes while soft limits are enabled on finite/homed axes

**Will this feature appear to a lot of users?**
GRBL-ESP32 is one of the few GRBL branches that offer more than 3 axis support. It's likely that many users consider this branch specifically for that support.

ExploratoryDevices, 2020-10-18:
Another, possibly quicker work around for this would be to have the "cold-start" Mpos be set to the home/Mpos value on boot.

bdring, 2020-10-18:
A soft limit mask has been requested before. I'll add that to our wish list section of our work tracking system.

The way the homing works with $H and single axis homes, there is no good way to do something with axes that are not homed.

ExploratoryDevices, 2020-10-19:
Thanks!

In the meantime, I set up user defined homing for my machine with this as the function:
`bool user_defined_homing() {
    float steps_per_mm = axis_settings[A_AXIS]->steps_per_mm->get();
    float travel = axis_settings[A_AXIS]->max_travel->get();
    sys_position[A_AXIS] = (travel * steps_per_mm) / 2;	// Set A axis to midpoint of travel to avoid soft limits.
    return false;
}`

Quick test shows that it seems to work.

bdring, 2020-10-19:
Great,

I probably would have done something similar.

I am going to close this. I have it on our radar for a Setting fix.

closed 2020-10-19 by bdring