bdring/Grbl_Esp32 #651: .ino is not compiling

giancape, 2020-10-24:
sketch\src\Motors\TrinamicDriver.cpp:22:24: fatal error: TMCStepper.h: No such file or directory
Multiple libraries were found for "WiFi.h"
compilation terminated.
 Used: C:\Users\gianc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Multiple libraries were found for "SD.h"
 Used: C:\Users\gianc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
Error compiling for board ESP32 Dev Module.

bdring, 2020-10-25:
Please follow this instructions on compiling the firmware.

https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware

Make sure you install the TMCStepper library per the instructions.

https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware#copy-libraries

giancape, 2020-10-26:
Thanks a lot for informations. My schede is ESP wroom32 now it
compile 

in machine,h there is include "Machines/test_drive.h 

select
option partition scheme minimal SPIFFS .etc etc......

But I have
contnuos reboot of esp32 every seconds.(only usb connected, nothing else
)

Bah !! don't worry about it I'll not use this Thanks 

Ciao

Il

> Please follow this instructions
on compiling the firmware. 
https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware [1] 

> Make sure you install the TMCStepper library per the instructions. 

https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware#copy-libraries
[2] 
thread.
unsubscribe [4].
  

Con Tiscali Mobile Smart 30 4G hai minuti illimitati, 100 SMS e 30 Giga in 4G a soli 8,99€ al mese. http://tisca.li/smart30

closed 2020-11-02 by bdring