bdring/Grbl_Esp32 #660: Setting Named Parameters in the Firmware

reynolds087, 2020-11-02:
Hi, sorry if this question is already answered somewhere in the documentation, but I'm having trouble finding it. I know that some of the named parameters such as max travel, max acceleration, etc. can be set in variables in the machine configuration file, so that if you load the firmware, the correct parameters will be set, but can all named parameters be set in the firmware? If so, I am looking for a list matching the named parameters on this page: [(https://github.com/bdring/Grbl_Esp32/wiki/Settings)]
and matching them to the variable name you can use in the config files for a `#define` statement, such as something like the following: `#define DEFAULT_X_MAX_TRAVEL 1000.0`

The problem I am trying to solve is that if I upgrade to a new version of the firmware I want to go through and make sure I set every parameter correctly, and the easiest way to do that is to use the configuration file, but some of the settings, I have to set in the console after flashing the unit because I don't know how to set them in the config file.

I suppose another useful way to accomplish this would be if you can import and export parameters after loading the firmware. If that is currently supported, can you please point me to the documentation?

bdring, 2020-11-02:
If you don't define a default yourself, it is generally pulled from the defaults.h file. That is a good place to look for them.

I was going to put that information on the Settings wiki page, but the table would get too wide. I have started a spreadsheet, but it is far from done. Soon they will be able to be imported along with everything else in a standard machine definition file. One compile for everyone.

BTW: While putting all your defaults in the machine definition is generally a good idea, they are **not lost** during an upgrade.

MitchBradley, 2020-11-02:
There is a "$Settings/ListChanged" command, which can be abbreviated as "$SC", that will show you every setting that is different from the default value.  Its output can be saved and replayed onto a fresh system.

In general, though, as Bart said, firmware upgrades do not lose setting values.  That feature was a primary design goal of the new named settings mechanism.  Prior to named settings, the storage format for settings was rigid and firmware upgrades were likely to invalidate it.  With named settings, the items are stored in a list that is tagged by the setting name, so they can be matched up across updates.

reynolds087, 2020-11-03:
Thanks, guys. That is very helpful. Appreciate the quick response.

closed 2020-11-04 by bdring