bdring/Grbl_Esp32 #661: Galvonometer control

MakerMatrix, 2020-11-03:
**Please describe the feature you would like implemented**
G-code control of scanning laser galvonometers.  I guess this would require some transformations that convert between cartesian and galvo DAC space.  If I've understood correctly, I've seen this implemented as a LUT that contains pre-sampled trig functions in units of effectively galvo voltage.  Then you just interpolate between them to get the correct voltage for any angle you need, so that it runs fast on lightweight hardware.  E.g:

https://github.com/obstruse/LaserDemo/blob/master/LaserDemo/Basics.cpp

**Why do you think this would improve Grbl_ESP32?**
It might make the laser spindle more useful for many projects.  You could build fast SLA or SLS 3D printers with that. ;-)

**What do you need the feature for?**
Building fast SLA or SLS 3D printers.

**Will this feature appear to a lot of users?**
Yes.  https://www.facebook.com/Photonsters

bdring, 2020-11-04:
I have a lot of experience with lasers and a little with Galvos. This would be a huge project, with a lot of firmware and hardware development and likely a lot of hardware costs. The internal DACs on the ESP32 are only 8 bit. PWM is likely a non starter with multi-kilohertz Galvos

Nothing like this is on our already very full roadmap. I would have to give a strong thumbs down on this at this time.

MakerMatrix, 2020-11-04:
In the little bit of research I've done, the way to drive galvos with the ESP32 is to hand the job off to an external DAC.  You drive a 16-bit DAC with the two I2S peripherals.  So I don't believe the hardware piece is terribly complicated.  Just convert the x value to a 16-bit number, send it to x, same for y.  There are hackaday projects built around that idea.  It would be a boon to get this functionality built into a full-fledged gcode interpreter.

bdring, 2020-11-04:
You are asking at a bad time. All devs are putting full time into higher priority features. We might be willing to listen next month.

MakerMatrix, 2020-11-04:
I am a patient man.  I might even be able to pitch in.  Thanks for considering it.

NairoDorian, 2021-05-21:
https://www.youtube.com/watch?v=_tizRU57rVE

This is awesome !!

Also, this man already made very clear tutorial to control galvo with external dacs with the esp32
https://www.youtube.com/watch?v=bl1e54QGJk4

This could be a game changer!

How can I help to implement this?

AlexBollmann, 2021-11-03:
Maybe it would be easier to use a digital galvo scanner with external controls. So you could use the XY2-100 protocol (which seems to be the standard of galvo control protocols...) 

[https://sigrok.org/wiki/Protocol_decoder:Xy2-100](https://sigrok.org/wiki/Protocol_decoder:Xy2-100)

Do you think this would be possible to implement?