bdring/Grbl_Esp32 #662: connection with grbl panel

qiwalker, 2020-11-03:
hi 

i try to connect with grbl panel
but getting error

is there any differend communication between grbl 1.1f with grbl_esp32 ?

thanks

MitchBradley, 2020-11-03:
The communication is intended to be fully compatible but Grbl_Esp32 has some extensions that might possibly cause problems.
When you report a problem, please include more detail.  It is impossible to debug a problem that is reported as simply "getting error".

MitchBradley, 2020-11-03:
I downloaded grbl panel and did a quick test.  After connecting, an error dialog popped up that said 
[image]
The data in the DRO boxes looked interesting
[image]

That suggests that grbl panel is failing to parse the status report correctly.  It is possible that grbl panel is trying to use grbl 0.9 parsing rules even thought the format is in grbl 1.1 style.  My guess is that grbl panel is trying to use startup messages to determine the status report version, instead of just looking at the report itself.

The GitHub site for grbl panel says
**This will be my last release, it has been fun and rewarding but the required enhancements that I think the program needs are beyond the effort I am able to expend.**
That means that grbl panel is no longer supported, and that the author knows it has problems.

closed 2020-11-03 by MitchBradley

qiwalker, 2020-11-06:
Thanks Now it work
thankyou for mentioning this
"My guess is that grbl panel is trying to use startup messages to determine the status report version, instead of just looking at the report itself."