bdring/Grbl_Esp32 #664: Problem A Axis

qiwalker, 2020-11-06:
hi 
i try to use several grbl software when first connection and jog A Axis "1mm" value on DRO never "1.000" it will "0.990"
for XYZ DRO working good

[image]

with arduino serial monitor

[image]

thanks

qiwalker, 2020-11-06:
with UGS

[image]

qiwalker, 2020-11-06:
with UGS

[image]

MitchBradley, 2020-11-06:
The A axis is rotary, calibrated in degrees not mm.  Furthermore it has to be converted to radians for use with trig functions. There is an additional conversion factor relating to the machine step size.  Those conversion factors, coupled with the finite precision of computer arithmetic, can result in roundoff errors. It is very common for DROs on machines to report values that are close to - within the mechanical precision of the machine - but not exactly equal to the requested value.  I seriously doubt that your A axis is accurate to within 1/100 of a degree.

qiwalker, 2020-11-07:
FYI
i made customachine and the A axis i use same like z axis function, im not use A axis as rotary but as linear sir, 
1. when i do homming (for the setting enable set force to origin) the dro for A axis read normally like xyz axis

2. when disconnected (esp32 cable still plug in pc) and reconnected, jog the

closed 2020-12-21 by qiwalker