bdring/Grbl_Esp32 #666: Anyone have problems with GPIO33,25,26,27,14 as control switch inputs?

stvcmhr, 2020-11-12:
I have probe, door, start, hold, and reset connected to GPIO33, 25, 26, 27, 14 respectively.  They are correctly called out in my machine definition file.  When running, grbl shows them as all active (i.e. low).  There is some signal conditioning circuitry (series 100 ohm resistor, 10K pullup, 100nf cap to ground).  When I check them with a voltmeter, they are low.  They aren't connected to switches yet, so that isn't the issue.  Is there a chance that they aren't being correctly configured as inputs? 

Any help would be appreciated.

Thanks,

Steve

bdring, 2020-11-12:
Do they switch states when activated?

If so, you can invert the logic in config.h

https://github.com/bdring/Grbl_Esp32/blob/ec5c95919f738a61399d518d21b61a324579817b/Grbl_Esp32/src/Config.h#L55

stvcmhr, 2020-11-12:
They do not appear to change state when I drive the pin high or low.  I can drive the pin high by connecting it to 3.3v with a low resistor (220 ohm).  I verify that it is high with a voltmeter.  However, grbl reports it as low.  

To be clear, I can drive the pin high or low, verified with a voltmeter, but grbl always reports them as low.

bdring, 2020-11-12:
Please paste your machine definition and startup text here.

https://github.com/bdring/Grbl_Esp32/wiki/Requesting-Help#boot-messages

stvcmhr, 2020-11-12:
I will post the machine definition, but I'm not sure what you mean by the startup text.

[grblESP32board.h.txt](https://github.com/bdring/Grbl_Esp32/files/5532596/grblESP32board.h.txt)

bdring, 2020-11-12:
The boot messages. Follow the link I provided.

stvcmhr, 2020-11-12:
I will grab the boot messages as soon as I get home.  I don't have access right now.

Thanks for the timely support!

Steve

bdring, 2020-11-12:
It looks like your control inputs are commented out.

These are all well tested and simple features. It is likely there is something wrong with your circuit or wiring.

stvcmhr, 2020-11-12:
I just commented them out last night to get the rest of the system going.  They were not commended out when I did the testing.  I agree there must be something wrong on my end.  I was just wondering if there was anything special about those pins, as they can be capacitive inputs.

I'll do more debug on my side before I waste any more of your time.  

Thanks for all the hard work,

Steve

bdring, 2020-11-12:
Some pins have some special features and restrictions. See this page.

https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins

stvcmhr, 2020-11-17:
Well, it turns out when you accidentally swap the 100 ohm series resistor with the 100nf parallel capacitor, things don't work so well.  Damn, those 0603 parts are tiny...

closed 2020-11-17 by bdring