bdring/Grbl_Esp32 #667: Sending Commands Over WiFi

reynolds087, 2020-11-13:
I am currently using ESP3D to control my CNC over WiFi and run my gcode programs from the SD card. I have noticed that Candle and a few other senders have added height mapping into their feature list, and I assume they are just sending probe commands to the CNC then recording the results into an array which they use to post process gcode files with arcs that follow the height map. I want to do something similar to this, but I don't see a way to do it over WiFi with ESP3D.

I am hoping you can provide some info about how to connect to the board over WiFi. I have some basic knowledge with networking and programming, and would like to write a sender program. I tried to telnet to the board on port 80, but that doesn't seem to work. How do I start a terminal session with the board over WiFi? I assume ESP3D is doing this on the back end?

MitchBradley, 2020-11-13:
Telnet is port 23.  Port 80 is HTTP

reynolds087, 2020-11-13:
That doesn't seem to be working for me. I do have a firewall that is a bit restrictive. But to be clear, I should be able to telnet to the board on port 23? In Putty, it opens the connection, but then it says "Connection Timed Out". I can get to ESP3D, but I suppose that doesn't rule out a firewall issue.

luc-github, 2020-11-13:
with putty you must use raw mode not telnet

reynolds087, 2020-11-13:
Thanks, Luc. So raw mode, port 23 should work then? Still not getting through.

luc-github, 2020-11-13:
check in your webui UI if port is set to 23 (it should be but in case)

reynolds087, 2020-11-13:
Thanks, that worked!

closed 2020-11-13 by MitchBradley