bdring/Grbl_Esp32 #671: Direction Issues

Rob-King-of-the-Plants, 2020-11-16:
I use my ESP32 with this cnc shield:
 https://www.az-delivery.de/en/products/az-delivery-cnc-shield-v3?_pos=3&_sid=4faba1877&_ss=r
and A4988 Drivers. 
Checking the pinout of the Arduino board this was meant for i wrote my own machine and wired everything respectively. 
However my motors dont seem to react to the direction signal. They always run in one direction until at some seemingly random moment they follow the correct direction. The Y axis seems to listen somewhat better. My suspicion was that the issue is caused by noise from the drivers, so i connected the the motors using shielded 4 wire telephone cable. This does not seem to make much of a difference though. I double checked correct wiring and crimping , could not find anything wrong there, 

Infos on my machine 
[MSG:Grbl_ESP32 Ver 1.3a Date 20201004]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESP32_V4]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:X  Axis Standard Stepper Step:GPIO(18) Dir:GPIO(19) Disable:None Limits(0.000,300.000)]
[MSG:Y  Axis Standard Stepper Step:GPIO(16) Dir:GPIO(17) Disable:None Limits(0.000,300.000)]
[MSG:Z  Axis Standard Stepper Step:GPIO(22) Dir:GPIO(23) Disable:None Limits(-300.000,0.000)]
[MSG:Global stepper disable pin:GPIO(13)]
[MSG:PWM spindle Output:GPIO(27), Enbl:GPIO(26), Dir:None, Freq:5000Hz, Res:13bits]

[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:X  Axis limit switch on pin GPIO(32)]
[MSG:Y  Axis limit switch on pin GPIO(33)]
[MSG:Z  Axis limit switch on pin GPIO(34)]

It might be something obvious to others, I am just starting out with all this. So you could save me a lot of time pointing me in the right direction.
Thank you very much in advance.

bdring, 2020-11-16:
I don't know much about that board or how you connected it.

Random direction or rough motor motion is a classic symptom of a wiring issue. See this in the wiki.

https://github.com/bdring/Grbl_Esp32/wiki/FAQ#motor-moves-erratically

Phone cord does not seem appropriate for stepper motors. You want heavier gauge multi strand wiring. 18-20awg.

Try with one motor driver installed at a time.

Rob-King-of-the-Plants, 2020-11-16:
Its a shield for the Arduino Uno R3. I covered every Pin on the bottom side of the  shield and on the ESP32 with Dupont connectors, fitting crimps where i needed a connection. 
With only one driver and one motor at a time Y still follows directions , Z follows directions most of the time but X still does not. I have continuity from the crimp on pin 19 to the crimp on the direction pin of the shield. I also have continuity from the direction pin of the shield to the direction pin of the driver. Furthermore all the pins on the shield where the motors are plugged in have continuity to the respective pins on the drivers.   I tested different motors out of the box so their connectors shouldnt be at fault.

[image]
[image]

MitchBradley, 2020-11-16:
Arduino have 5V signaling; ESP32 has 3.3V.  Perhaps the shield is reacting badly to the voltage mismatch.

MitchBradley, 2020-11-16:
Get a voltmeter and check the voltage between GND and XDIR.  Enter these commands
```
G91
G1 X10 F30
G1 X-10
```
During the first G1 command - which should take 20 seconds to finish, the XDIR voltage should be either 0V or 3.3V, and during the second G1 command, the XDIR voltage should be the opposite from what it was during the first G1.

If that does not happen, either there is a wiring problem between the ESP32 and the shield, or the firmware configuration does not match the ESP32 pin that you are using for XDIR.

Rob-King-of-the-Plants, 2020-11-16:
I checked this on direction pin on the shield and afterwards to make sure that it  isnt the wiring directly on the ESP pins. 
Z behaved as expected but X was the whole time on 3-3.3V (no change between shield and ESP) , as seen in the pictures.

[image]
[image]

MitchBradley, 2020-11-16:
Which ESP32 pin are you using for XDIR, and what is the contents of the machine configuration file.

Rob-King-of-the-Plants, 2020-11-16:
#define X_STEP_PIN              GPIO_NUM_18
#define X_DIRECTION_PIN         GPIO_NUM_19

You mean my .h file? The pins referred to above  are used for X. 

However thanks to the double checking both of your input made me do i came to realize that the only thing left which could be faulty was the ESP32. I unpacked a fresh ESP32 compiled GRBL and now it seems to work. I hope it just comes down to inexperience with Arduino IDE. I came to all this by working with micro python on the ESP32s. 
I suspect that my ESP32 isnt properly flashed and confused by older iterations. The other thing might be that i damaged it by accident or it came damaged to me. 
I am going to check the faulty ESP32 in webrepl hoping it just was the flashing. 
Thank you both for your time.

closed 2020-11-16 by Rob-King-of-the-Plants