bdring/Grbl_Esp32 #672: Guide to use for production

rijulmanch, 2020-11-18:
I run a small manufacturing shop and want to retrofit 3 old milling machines on which I do some production runs (ie. make lots of 30-50 pieces in one go)
I have a variety of products (50 variants) for which I wish to store the G-code in SD card. Now Question is -
1. How do I repeat (loop) same code with push of a button.
2. Is there an easier way to select the product, like scanning a barcode? I want to make it dummy proof
3. How do I error proof the correct board selection? For example A physical USB serial connection would be great for dummy proofing.

bdring, 2020-11-18:
None of this is in the firmware. It is too specific to one machine.

There are ways to add custom features. This could as simple as a custom behavior after a button push or as complex as the kinematics for a delta robot. You do this by adding a custom code file. There are several examples in the [Custom folder](https://github.com/bdring/Grbl_Esp32/tree/main/Grbl_Esp32/Custom).

MitchBradley, 2020-11-18:
I am sympathetic to your needs, as I have done things like this in my own workshop, for machines that I have deployed at vendors, and for product test rigs that I have deployed at factories in China - where you cannot assume that the operators have more than the bare minimum of training and cannot read English.

Regarding 1 and 2, it would seem to me that 2 sort of takes care of 1, since re-scanning the barcode is not so different from pushing a button.  In checkout lines, when faced with a small number of similar items, the common gesture is just to pass the same item over the scanner multiple times.

I'm not sure what you mean by 3, unless the idea is that you are using the same UI surface for all the machines - which seems like a bad idea in a production shop.  If you have a UI surface per machine, then you can indeed connect each to its controller via a dedicated USB.  That is already supported.  Alternatively, you could use bluetooth and pair each control surface with the associated controller.

So let's talk about 2 - barcode scanning.  You can print barcodes with any text that you want, including commands to run specific SD files ($SD/Run=filename). If the length is a issue, use short filenames like AA, or use a 2D barcode - at the expense of a slightly more expensive scanner - about $10 to $20 more.  So what about the hardware connection?  USB does not work well in this case because the USB connection on the ESP32 modules is the target end not the host end, and the USB barcode scanners are intended to plug into a host.  But there are Bluetooth scanners!  Some of them offer Bluetooth SPP mode, which can talk to the ESP32's bluetooth port. Here's one that looks good: https://scanavenger.com/product/scanavenger-wireless-barcode-scanner/

MitchBradley, 2020-11-18:
This one is less expensive and, according to the rep, has the same engine.  https://scanavenger.com/product/sa9100-series-pro-wireless-bluetooth-barcode-scanner-1d2d-with-vibration-extended-range-ultra-fast/

MitchBradley, 2020-11-18:
The above referenced products have the ability to set a customized prefix up to 32 characters, so you could put the $SD/Run= prefix in the scanner in order to reduce the length of a 1D barcode - then print the barcodes with a label printer.  Or you could just use a long 1D barcode, or a 2D barcode.  If you arrange for one of them to magically appear at my doorstep, I will help with system integration.