bdring/Grbl_Esp32 #674: dual axis machine, dir mask

manfredoo, 2020-11-19:
Hi everyone,

I'm trying to get my XYYZ machine going but have a problem with the dual axis.
My servo drives don't seem to be able to change direction of rotation on their own so I have to give the bit mask a go.
However I can't get my head around on how I should do that:
I assume it should still be a 3 axis machine in config.h, but the bit mask only speaks about x, y and z.
So what do I do if I want one y axis to turn in one direction and the other y axis opposing to the first one?

I configured my machine.h file as following:

#define USE_GANGED_AXES // allow two motors on an axis

#define X_STEP_PIN      GPIO_NUM_12
#define Y_STEP_PIN      GPIO_NUM_14
#define Y2_STEP_PIN     GPIO_NUM_21  // ganged motor
#define Y_AXIS_SQUARING
#define Z_STEP_PIN      GPIO_NUM_27

#define X_DIRECTION_PIN GPIO_NUM_26
#define Y_DIRECTION_PIN GPIO_NUM_25
#define Y2_DIRECTION_PIN GPIO_NUM_22 // ganged motor
#define Z_DIRECTION_PIN GPIO_NUM_33

Thank you all

bdring, 2020-11-19:
`#define Y_AXIS_SQUARING` is no longer used. Dual motor is inferred from the 2 motor definitions. 

Squaring is set by the **$Homing/Squared=Y** Setting

Do you the pinout of your servos? Are they compatible with 3.3V signals?

manfredoo, 2020-11-19:
Yes they are compatible with 3.3V, X and Z axis are working fine. Both Y axis are working aswell, other than not being in opposing directions.
I'll remove Y_AXIS_SQUARING tomorrow and see how that goes.

Thank you for your help

bdring, 2020-11-19:
Do both Y motors move and reverse direction? Is the problem they don't move in opposite directions?

If so, this currently cannot be fixed in firmware. It can easily be fixed with stepper motors. You just swap the wires on one coil on one motor.

You might be able to change active low and active high on the driver input. Can you send a link or picture of your driver?

manfredoo, 2020-11-19:
Yes, the only problem is the same direction of both motors.

I can't change the motor coils since these are AC servos with encoders [https://bit.ly/3nIAYVx].
I missed that the drivers can only change motor direction by setting in speed mode but not in position mode which I am using.
I'll try changing active low/active high tomorrow too.

Since I basically have very very limited programming skills, would it be possible to invert Y2 direction signal by using the ~symbol? like:
#define Y_DIRECTION_PIN GPIO_NUM_25
#define Y2_DIRECTION_PIN ~GPIO_NUM_22

bdring, 2020-11-19:
You could certainly hack it. 

We are working on the next generation of the firmware and how pins are defined and used. We don't want to muddy things by hacking temporary fixes. 

 Can you send a link or picture of your driver?

manfredoo, 2020-11-19:
Here is a file of the driver, I hope the upload is working out.
[LICHUAN B2 Servo Driver Instruction Book(1).pdf](https://github.com/bdring/Grbl_Esp32/files/5569492/LICHUAN.B2.Servo.Driver.Instruction.Book.1.pdf)

bdring, 2020-11-19:
Is (19) a method of reversing direction one of them? Hard to tell with the wording.

[image]

manfredoo, 2020-11-20:
The most right column says 'T', which refers to torque mode, I'd need a 'P' for position mode there.
Thank you so much for digging this deep into my issue.

bdring, 2020-11-20:
How are you wiring these pins.  

[image]

I think we can flip the direction here.

bdring, 2020-11-20:
I think you can wire this circuit 2 ways.

[image]

Try wiring one way for each driver. Vcc should be 3.3V

[image]

manfredoo, 2020-11-20:
Looks like the best scenario I could do, thank you very much for suggesting that.

bdring, 2020-11-20:
Let me know if it works.

manfredoo, 2020-11-20:
Hi,
I just changed the wiring for the Y2 axis and it's working perfect now. So thank you very much for your time and help.

closed 2020-11-20 by manfredoo