bdring/Grbl_Esp32 #675: Telnet connection using only CR as delimiter does not process commands

HuubBuis, 2020-11-20:
What version of the firmware are you using? 1.3a

Is the problem repeatable? yes

Under what conditions does the bug occur?  telnet connection using only CR as delimiter

Commands like $$ and Gcodes are not processed, immediate commands like ? are processed.
In serial and Bluetooth connections, commands using only CR as delimiter are processed!

The "bug" can be tested using putty and the following settings. 
- extra putty | setting | End Of Line   set to CR
- Connection | Telnet |  disable Keyboard send Telnet special commands
- Connection | Telnet |  disable Return sends Telnet New line instead of ^M

I have fixed it in my GUI so for me it is not a problem anymore. It took me a very long time to figure out what was going wrong.
I prefer the CR as a delimiter, fewer characters to send occupying less buffer space.

If the delimiter is set to CRLF, all is working OK.

bdring, 2020-11-21:
I found that it is only using LF as the delimiter. I'll look into why CR is not working.

bdring, 2020-11-22:
CR was filtered out to be more compliant with the telnet spec. We are discussing in the development forum  what is the best solution.

HuubBuis, 2020-11-22:
I don't mind the delimiter, had just totally not expected there would be a difference between sending using serial, Bluetooth or WiFi. I thought there had to be a timing problem in my GUI! 

I doubt between Lf that is telnet compliant and accepted by grbl as well or CrLf to avoid problems on future controllers.

bdring, 2020-11-22:
Change has been merged to devt branch

closed 2020-11-22 by bdring