bdring/Grbl_Esp32 #681: Macro Button Proposal - Request For Comments

bdring, 2020-11-24:
### Macro Button Proposal - Request For Comments

We currently support 2 types of macros.

-  **WebUI**: You can upload a gcode file to the Flash (not SD) storage on the ESP32. You can then create a button for the WebUI and assign a file to run when pushed.
- **User Macro Buttons**: These are 1 to 4 physical buttons wired to the controller. When the button is pushed it executes a function that the user provides. This can do anything from running static gcode or procedural, "if this then that" type things.

Macro buttons are a commonly requested feature, but the coding requirement scares some people off. The proposal is to have a default, basic mode where physical buttons act like the WebUI and an advanced mode where you provide functions. There would be Settings like **$User/Macro0=**  that would tell the button what to do.

The question is what the setting should provide.

- **A line(s) of Gcode?** (limited in length)
- **A file to run from the SD card?** (these files might be a little transient)
- **A file to run from the NVS (Flash) of the ESP32** (more stable, but currently you need the WebUI to upload)
- **Hybrid with some sort of prefix** (possibly confusing)
  - **GC:G53G0Z-1** (for gcode)
  - **NVS:\foo.nc** (to run file /foo.nc from the ESP32)
  - **SD::\bar.nc** (to run /bar.nc from the SD card)

Is the setting name **$User/Macro0=** good or is there a better name?

Thoughts...comments?

MitchBradley, 2020-11-24:
We have the existing commands $LocalFS/Run=filename  and $SD/Run=filename .  Maybe the value of the Macro setting is just a verbatim command.

$User/Macro0=G20
$User/Macro1=$SD/Run=foo.nc
$User/Macro2=$SD/Run=bar.nc

MitchBradley, 2020-11-24:
For verbatim GCode, we could use & as a line separator, translating that into a newline before handing it off.  I'm unaware of any existing use of & in GCode - although I haven't done an exhaustive search.

bdring, 2020-11-24:
I like that. A simple replace before doing anything else is probably easiest.

HuubBuis, 2020-11-24:
I would prefer a "command file" to execute like Command0.txt .. Command4.txt.
The content of the file could be a line of gcode, the name of a file to execute, etc. Editing a text file is within the capabilities of most users. In time, the format/language can be extended for future needs.

bdring, 2020-11-24:
@HuubBuis That is one of the options @MitchBradley is talking about. foo.nc and bar.nc are both command files. One is stored on the ESP32 and one is on the SD card.

I tweaked the code and these 4 macros work fine.

```
$User/Macro3=$LocalFS/Run=upg30.nc
$User/Macro2=$SD/Run=upg30.nc
$User/Macro1=G53G0Z-1&G28
$User/Macro0=G28
```

HuubBuis, 2020-11-29:
I have come up with another useful (I think) option, a "macro" to send a (push) message to the GUI, so the GUI can response to buttons.

rijulmanch, 2020-12-05:
Is it possible to run a function like this and run it from within the GCode and also with a Macro button?
 
void sequencerelays(){
              digitalWrite(E1, HIGH);
              delay(500);
              digitalWrite(E2, HIGH); 
              delay(200);
              digitalWrite(E2, LOW);
              delay(200);
              digitalWrite(E1, LOW);
              delay(500);
   }

bdring, 2020-12-05:
Yes,

**Via GCode** You can already do this. You would [setup (2) digital I/O pins](https://github.com/bdring/Grbl_Esp32/wiki/M62,-M63,-M64,-M65-&-M67-User-I-O-Commands) and control them with the following gcode.

```
M62P0
G4P0.5
M62P1
G4P0.5
M63P0
G4P0.5
M63P1
G4P0.5
```

**Via Button Macro** Setup those same pins and create the macro below, which is the same as the gcode with the & character used as the line end.
```
$User/Macro0=M62P0&G4P0.5&M62P1&G4P0.5&M63P0&G4P0.5&M63P1&G4P0.5
```

rijulmanch, 2020-12-07:
Actually, I can think of longer sequences like this which are used multiple (20-30) times within one gcode. 
Would be very handy to just call a function.

sjonholle, 2020-12-09:
I had to remove all the other files in the Custom directory because it kept saying it couldn't find etc/settings.h in coreXY.cpp. Only my own costum file in this directory and i control my vacuumcleaner (great)