bdring/Grbl_Esp32 #682: Add External Driver

liwajaya, 2020-11-26:
hello everyone who understands this, 
please enlighten me whether the scheme for adding an external [stepper driver dm542 ](https://robokits.co.in/motor-drives-drivers/stepper-motor-driver/dm542-digital-microstepping-stepper-driver-4.2a-leadshine-original)like this is correct,

[image]

thanks

MitchBradley, 2020-11-26:
That will work.  The polarity will be inverted with that wiring, but the software can be configured so that is okay.

Another way that will work is to connect the - terminals on the external driver to ground, and drive the + terminals from the 74AHCT595 outputs.

The external adapter https://www.tindie.com/products/33366583/external-stepper-motor-driver-adapter/ allows either approach.

joedirium, 2020-11-26:
I am using the recommended adapters with DM542 drivers with jumper set to common ground and it is performing pretty well. Only issue for me so far is that the enable signal is not set properly and I cannot set and save the invert flags. I will have to compile a newer version which will probably fix this.
Just one recommendation:
running DM542 with 24V is possible, but from my understanding 48V would be the better setup if you are free to choose.
I am using 36V, just because I had such power supply left over.

bdring, 2020-11-26:
@liwajaya As @MitchBradley says, either way will work. The opto input on those inverts the signal again. some drivers will prefer rising edge or falling edge. Your driver has a way to set that.

bdring, 2020-11-26:
@joedirium What controller are you using?

joedirium, 2020-11-26:
I am using
the ESP32 sixpack board
5 x "external driver connector v0.5"
5 x DM542 drivers.

Connected are
3 x Nema23
2 x Nema17 motors.

The DM542 do not allow a lot of configuration anyway:
SW1 to SW3 set the current 
SW4 to SW8 set the steps per rev (microsteps)

bdring, 2020-11-26:
What do you mean you can't save the invert flag. Does it not save or not function properly?

joedirium, 2020-11-26:
I does not save the values - always defaults back to 0.
But as mentioned, I will have to compile a later version. Mine is already a few months old. If this problem persists I will open a ticket. In the meantime I can simply ground the enable input of the drivers.

closed 2020-11-29 by bdring