bdring/Grbl_Esp32 #687: Auto square endstop offsets

RootCNC, 2020-12-01:
*I might be mistaken but I cannot find information on it*

**Please describe the feature you would like implemented**
Request the addition of the option to apply software offset for fixed endstop on cloned axis. For example, the Root CNC and MPCNC has a gantry style system with dual axis motors. Doing a home routing ($H) I believe the software assume the error has been removed by mechanically adjusting the endstop in order to yield a square machine. It would be amazing if GRBL ESP32 can feature an offset to for a given axis. Similar to Reprap Marlin firmware and also this fork of grbl-Mega-5X,

Marlin information [LINK](https://marlinfw.org/docs/gcode/M666-dual.html)
V1 forum post [LINK](https://forum.v1engineering.com/t/grbl-on-rambo-w-dual-endstops/10981)

**Why do you think this would improve Grbl_ESP32?**
I believe so - this will allow for fine control of gantry racking.

**What do you need the feature for?**
Gantry style machine that are controlled by two motors.

**Will this feature appear to a lot of users?**
I hope so - Root CNC and MPCNC has a number of build world wide. 

Appreciate the support!

bdring, 2020-12-01:
I'll add this to the "wishlist" section of our development tracking system. I can see how it might be easier to integrate switches into parts that don't move on some machines.

RootCNC, 2020-12-01:
Thanks Bart,

Unfortunately, the endstops have been fixed on the Root CNC machine right from the earliest  - you can see them [HERE](https://i2.wp.com/rootcnc.com/wp-content/uploads/2020/09/ex_20200404_183659-scaled.jpg?fit=485%2C1024&ssl=1). They're not placed in this photo but I hope you can see the position the endstop would be mounted in.

Thanks again!

reynolds087, 2020-12-01:
The limit switches/mounting brackets I have are very cumbersome to adjust, and don't have any fine tuning mechanism. Being able to define the offset in software would greatly increase the dimensional accuracy of my gantry. I ended up purchasing some hall effect sensors and designed a bracket to adjust the magnet distance with a screw, but it required a redesign of my control board, and still not fully implemented.

Here is a second vote for this request.

RootCNC, 2021-02-12:
Is there any updates on this?

bdring, 2021-02-12:
We have not done anything on this yet.

We are working on 2 versions of the firmware. 

- With the current version, we are fixing issues as they come up and adding only critical features.
- We are also working on a next generation version where it is much easier to add complex settings, like a limit switch with multiple parameters.  The primary feature of the next generation is the configuration is done at run time. You can completely change your machine without a recompile.

We don't want to have to add features to the current version when we know we would have to start over to do it a better way on the next version.

Our development roadmap is on a Trello board. This request is on there.

RootCNC, 2021-02-15:
Oh wow, fair enough. I totally understand.

I'll look into the Trello Board and keep and eye on it. 

Keep up the great work!