bdring/Grbl_Esp32 #690: New features and questions

PhilippEBZ, 2020-12-04:
Hello together!

Great project, I got it compiled onto a ESP32 I had around and got the motor to spin and got the gcode to play from the SD card! Looks promising!

But while playing with it I came up with the ideas / questions and hope someone can answer:

- would it be possible to have a visualization of the toolpath in the webinterface? Like in cncjs this is done on the client side - so it would not mean higher load on the CPU, or is that not possible with ESP32?  

- is there a possibility to connect a handpanel for jogging? I could imagine like 6 inputs and a poti for speed, or another ESP32 / teensy connected via i2c or bluetooth!?

- after flashing and reset i get like 150-200 times: 
    [E] [WiFi.Udp.cpp:219] parsePacket(): could not receive data: 9
and a couple of time while being connected to wifi:
    [E] [WiFiClient.cpp:392] write(): fail on fd 59 / 69, errno: 104, "Connection reset by peer"
   
and some other errors. 

Can anyone tell what I am doing wrong?

Greets
Philipp

[image]

MitchBradley, 2020-12-04:
In the future, please create separate tickets for each individual item.  Lumping them all together makes it difficult to focus.

> would it be possible to have a visualization of the toolpath in the webinterface? Like in cncjs this is done on the client side - so it would not mean higher load on the CPU, or is that not possible with ESP32?

The WebUI is a separate project at https://github.com/luc-github/ESP3D-WEBUI , so an additions to that would not be done at the Grbl_Esp32 level.  That said, I am working on a background project to add such visualization as a WebUI extension.  It is pretty far down on my ToDo list right now, though.

> is there a possibility to connect a handpanel for jogging? I could imagine like 6 inputs and a poti for speed, or another ESP32 / teensy connected via i2c or bluetooth!?

You can use an Android phone or tablet with this app: https://play.google.com/store/apps/details?id=in.co.gorest.grblcontroller

Small 7" tablets are inexpensive.

Another possibility is this: https://github.com/MitchBradley/grbl_controller_esp32/tree/GrblTCP . It uses hardware from https://www.makerfr.com/en/categorie-produit/cnc/ .  My fork of that software connects to Grbl_Esp32 over telnet

MitchBradley, 2020-12-04:
> [E] [WiFi.Udp.cpp:219] parsePacket(): could not receive data: 9

This is just a guess, but I suspect that something on your network is flooding Grbl_Esp32 with requests that are unrelated to Grbl.  Perhaps SSDP (service discovery) or OTA (over the air update) or something else.  You could try disabling some features by commenting out some of the lines in this block of Config.h

```
#define ENABLE_HTTP                //enable HTTP and all related services
#define ENABLE_OTA                 //enable OTA
#define ENABLE_TELNET              //enable telnet
#define ENABLE_TELNET_WELCOME_MSG  //display welcome string when connect to telnet
#define ENABLE_MDNS                //enable mDNS discovery
#define ENABLE_SSDP                //enable UPNP discovery
#define ENABLE_NOTIFICATIONS       //enable notifications
```

PhilippEBZ, 2020-12-06:
Hi, 

thanks for your answers! I saw in the wiki that there are already specifications for such a visualization. Nice!! Looking forward to that. I guess to actually use the webGUI  I can't comment out HTTP, but I will try to compile without OTA, SSDP and others and report here if that changes.

Regarding the jogging I think I was not specific enough. I imageine a jogging controller solution with physical buttons as I want to watch the machine and not my fingers when moving it. This might just be me, but in my job I move around some 3-4m tall robots, and I don`t like the idea of moving things with touchscreen, where I am not able to watch the machine.
So I guess the second suggestion with GrblTCP or over Telnet might be the best choice for me. I will check those options out!

Thank you!!!