bdring/Grbl_Esp32 #693: Uploaded Test Drive, connects to network but I can't connect to ESP3D WebUI via http://grblesp.local/

JonAllen816, 2020-12-04:
Bart, Thanks for keeping up the great work!

I have some experience with Grbl_Esp32 running on MPCNC_V1P2 and am curious to test drive the new WebUI and 6-pack controller.

Currently I don't have my ESP32 connected to anything except via USB to MacbookPro. I successfully compiled, uploaded and can monitor via VSCode/Platformio/Terminal but when I try to connect via hostname:http://grblesp.local/ or directly to 192.168.21 to initiate ESP3D/WebUI setup; the browser never connects and then the ESP32 is non-responsive via Terminal ([ESP401], [ESP420], [ESP444], etc.) until I reboot it manually.

This is my serial monitor output after re-connecting:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Grbl_ESP32 Ver 1.3a Date 20201124]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:Test Drive - Demo Only No I/O!]
[MSG:Axis count 3]
[MSG:Timed Steps]
[MSG:Init Motors]
[MSG:No spindle]

[MSG:Client Started]
[MSG:Connecting Lucky7]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connected with 192.168.1.21]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]

Grbl 1.3a ['$' for help]
[ESP420]
Chip ID: 27814
CPU Frequency: 240Mhz
CPU Temperature: 38.9C
Free memory: 109.03 KB
SDK: v3.2.3-14-gd3e562907
Flash Size: 4.00 MB
Baud rate: 115200
Sleep mode: Modem
Available Size for update: 1.87 MB
Available Size for SPIFFS: 169.38 KB
Web port: 80
Data port: 23
Hostname: grblesp
Current WiFi Mode: STA (30:AE:A4:DD:A6:6C)
Connected to: Lucky7
Signal: 82%
Phy Mode: 11n
Channel: 6
IP Mode: DHCP
IP: 192.168.1.21
Gateway: 192.168.1.1
Mask: 255.255.255.0
DNS: 192.168.1.1
Disabled Mode: AP (30:AE:A4:DD:A6:6D)
Current BT Mode: Off
Notifications: Disabled
FW version: 1.3a (20201124) (ESP32)

bdring, 2020-12-04:
Are you connecting to the WiFi access point it creates or using your home router?

JonAllen816, 2020-12-04:
I initially tried to connect in AP mode but couldn't get it to establish a wifi connection directly to GrblEsp so then I set it in STA mode which connected to my LAN.

192.168.21 is the IP of the ESP32, no typo; message above was copied from Terminal and I verified it with NETGEAR router manager: attached devices.

Wireless Devices (Wireless intruders also show up here)
#IP Address                           Device Name                         MAC Address
1192.168.1.21                                   --                               30:AE:A4:DD:A6:6C

I alternatively tried building/compiling/uploading again with platformio.ini :
[common]
build_flags = 
	-DMACHINE_FILENAME=4axis_external_driver.h
	-DCORE_DEBUG_LEVEL=0
	-Wno-unused-variable
	-Wno-unused-function

but got the same results; as soon as I try to connect to hostname:http://grblesp.local/ or directly to 192.168.21 it bricks the ESP32.

bdring, 2020-12-04:
"192.168.21 is the IP of the ESP32, no typo;"??

Try 192.168.1.21

Do you mean reboot or brick?

bdring, 2020-12-04:
Before doing anything with the WIFI try issuing this command..

[ESP710]FORMAT

This will reformat the SPIFFS. This can get corrupted if different types of upload were attempted.

Then power cycle the ESP32.

JonAllen816, 2020-12-04:
my bad, they were typos - I have been trying 192.168.1.21

probably not using brick correctly - it goes non-responsive

OR should I say WENT non-responsive... [ESP710]FORMAT -> Power Cycle -> http://grblesp.local/ -> ESP3D -> Load Index.html.gz = SUCCESS!

You're the man Bart! Thanks a lot buddy!

Now if I wanted to modify my MPCNC_V1p3 ;) (I made a copy of your MPCNC_V1p2 board but modified it to work with TMC2208's) to work with a 4th axis; and I cut the trace that connects Y_Dir to Y-B and jumped a wire to Y-B_Dir from open GPIO(0) to use as A_DIR? and then use Y_STEP_B (GPIO(21) as A_STEP? Would this work or do you see any obvious problems I'm missing?

bdring, 2020-12-05:
The issue you had is [in the FAQ](https://github.com/bdring/Grbl_Esp32/wiki/FAQ#trying-to-access-webui-crashes-esp32). It is rare. @luc-github might have a better explanation of the cause.

Your MPCNC mod might work. GPIO_NUM_0 is used to trigger the boot loader. You should see if it affects the ability to upload firmware.

Closing issue.

closed 2020-12-05 by bdring

luc-github, 2020-12-05:
FS corruption due to partition change is a known issue and description in FAQ is correct - it is ESP32 core / SPIFFS not in FW  -that said I have some ideas to contain the crash / dead loop  and  it is my todo list, but because it is an edge case and busy with 3.0  it is not in the top yet