bdring/Grbl_Esp32 #695: v3.5 board, Slack invite?

thestumbler, 2020-12-05:
My client has a few of the v3.5 3-axis boards, and realized that the header file in `Machines` has been removed for this board -- only v4 header file remains.  I couldn't track down the original version.  We temporarily rang-out the connections and made a new header file.  Is there some reason the 3.5 header file was removed?  I'm talking about `3axis_v3p5.h` (I'm guessing the file name).

Also, the project readme says to request an invite to join the Slack channel.  How? I can't find an email or mention of the Slack channel name.

Thanks,
-Chris

bdring, 2020-12-05:
We have some automated tests that run through all of the machine definitions. It uses Travis CI. The tests were taking too long and failing, so we trimmed out some of the older ones.

[The schematic is here](https://github.com/bdring/Grbl_ESP32_Development_Controller/blob/master/source/V3p5/schematic_esp32_cnc_test_v3p5.pdf)

Here is a [Slack invite link](https://join.slack.com/t/buildlog/shared_invite/zt-7twbithi-bRGScOaCgJoBpTbkykfxoA) I don't like to leave these public, so it will be removed after I see you join or in 24hrs

closed 2020-12-05 by bdring

thestumbler, 2020-12-05:
Thanks.  If anyone else needs this file, it can be retrieved from the repo history.  Last time it existed was commit 1774c50, so therefore...

`git checkout 1774c50 -- Grbl_Esp32/src/Machines/3axis_v3.h`