bdring/Grbl_Esp32 #696: Macro0 and Macro1 don't work with Feed_Hold.

Spla-sh, 2020-12-06:
Hello ! 
When I use Feed_Hold with Macro0 and Macro1, Grbl crash. 
With WEBUI, I can see when I push the Feed_Hold button. But when I push the Macro0 or Macro1 button, WEBUI stop receving messages and I have "connection error" message.
After reseting the ESP32, I disconnect the Feed_Hold pin and I can see "Pn:H" on WEBUI. 
After that I can push MC0 and MC1 without problems and see "PN:H,MC0,MC1"
I use 34-39 GPIO with pullup resistors.
Thanks !

bdring, 2020-12-06:
This is hard to follow. Please list the sequence of events, on separate lines, that causes the problem. 
- When you talk about a button make sure you say whether it is a button on the WebUI page, or a physical button.
- If it is a physical button, show the definition from your machine definition file
- If there is a macro, please give the complete text of the macro.
- Please give the machine state (Idle,Run,Hold,etc) when you push each button) 
- For each physical button please show the status command ("?" character) response for each button in the pushed and unpushed state.
- What firmware branch and version date are you using.

Spla-sh, 2020-12-06:
Sorry, hope this help : 

- It's 4 physical buttons (switchs)

- Def file 
`#ifdef CONTROL_SW_DEBOUNCE_PERIOD
    #undef CONTROL_SW_DEBOUNCE_PERIOD
#endif
#define CONTROL_SW_DEBOUNCE_PERIOD 100 // really long debounce

#ifdef INVERT_CONTROL_PIN_MASK
    #undef INVERT_CONTROL_PIN_MASK
#endif
#define INVERT_CONTROL_PIN_MASK B11111111 

#define CONTROL_FEED_HOLD_PIN   GPIO_NUM_36
#define CONTROL_CYCLE_START_PIN GPIO_NUM_39

#define MACRO_BUTTON_0_PIN		GPIO_NUM_34
#define MACRO_BUTTON_1_PIN		GPIO_NUM_35`

- I don't have created the cpp file for the macro yet.

- Machine state : 
 "Hold" when I push the hold button
"Idle" when I push the start button
"connection error" when I push the Macro 0 or Macro 1 button

- All the buttons don't show nothing in unpushed state. When pushed, Hold = "Pn:H" ; Start = "Pn:S" but Macro 0 and Macro 1 don't show nothing because Grbl don't respond anymore.

- I use the latest release (24/11/2020) of the main branch.

When I disconnect the Hold pin (GPIO36), I see "Pn:H" and I when I push the macro0 and 1, I can see "Pn:HMC0MC1"

Thanks !

bdring, 2020-12-06:
What do you want Macro 0 and Macro 1 to do?

Spla-sh, 2020-12-06:
Grab a file from the sd card like your polarcoaster

bdring, 2020-12-06:
I would suggest trying the devt branch. It has the features [described here](https://github.com/bdring/Grbl_Esp32/wiki/Macro-Button-Feature#doc-for-new-version-coming-soon).

Let me know if that works for you.

Spla-sh, 2020-12-06:
Yes it works ! When MC0 and MC1 buttons are pushed, I can see "Pn:01" while my "hold" gpio pin is connected.
Now I can make my cpp file for macros.
Thanks a lot !

closed 2020-12-07 by bdring