bdring/Grbl_Esp32 #703: ESP32S with bluetooth

manisaranbala, 2020-12-11:
My ESP32 chip is DOWDQ6 
FIRMEWARE UPLOADING GOOD BUT NOT SHOW BLUETOOTH 
HOW TO CONFIG BLUETOOTH ?

HuubBuis, 2020-12-11:
You also need to configure some settings using WiFi or serial. 
There is chapter in the WiKi explaining howto setup bluetooth [https://github.com/bdring/Grbl_Esp32/wiki/Using-Bluetooth](url)

manisaranbala, 2020-12-12:
[image]

ORIGINAL FILE

manisaranbala, 2020-12-12:
[image]
 I CHANGE  const int DEFAULT_RADIO_MODE = ESP_BT ;
YES WORKING GOOD THANKS SIR

bdring, 2020-12-12:
**$Radio/Mode=BT** will change it with the default compile.

...closing issue

closed 2020-12-12 by bdring