bdring/Grbl_Esp32 #710: 3 AXIS NOT WORKING

manisaranbala, 2020-12-18:
I NEED 3 axis step and dir pinout firmware AND circuit schematic

bdring, 2020-12-18:
You need to be more clear about what hardware you have and what version of the firmware you are using.

manisaranbala, 2020-12-18:
i used board esp32s and bluetooth via  connect GRBL Controller + android app
first i try Grbl_Esp32-main and upload successful  only change [enable bluetooth] uploading good then connect tb6600 stepper drive not working 
[image]

bdring, 2020-12-18:
Did you follow this instruction when compiling?

https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware#customize-important

You probably need this machine definition file.

Machines/3axis_v4.h

manisaranbala, 2020-12-18:
[image]

bdring, 2020-12-18:
**Test Drive - Demo Only No I/O**

That will not output any signals

Please follow the Wiki.

manisaranbala, 2020-12-18:
[image]

manisaranbala, 2020-12-18:
YES WORKING GOOD THANK YOU

bdring, 2020-12-18:
If my time has helped you, please consider donating to or sponsoring this project.

Closing...

closed 2020-12-18 by bdring

manisaranbala, 2020-12-18:
https://user-images.githubusercontent.com/5657183/102635423-24549980-4108-11eb-9c38-ea5617558c2a.mp4

manisaranbala, 2020-12-18:
Definitely leave Finishing Work

bdring, 2020-12-18:
YOU MUST CHANGE YOUR THIS

#include "Machines/test_drive.h"

TO THIS!

#include "Machines/3axis_v4.hh"

IN THE Machines.h FILE