bdring/Grbl_Esp32 #711: adding A Axis to core XY

qiwalker, 2020-12-18:
hi, is it possible to use A Axis in coreXY machine

thanks ^^

bdring, 2020-12-18:
Currently you are limited to 3 axes. It should not be too difficult to allow the use additional axes as long as the CoreXY remain on XY. 

I am looking at CoreXY this week. I will see if it is easy to add the extra axes at this time.

Symeth, 2020-12-19:
hi,

thankyou for ur reply,

that's good! ^^
yes main coreXY still on XY
i'll be wait, hopefully extra axis can be add in corexy machine

anyway i can't use homing on coreXY
already add those script on my machine config and enable homing in machine settings but still not working

#define USE_CUSTOM_HOMING
#define DEFAULT_HOMING_CYCLE_0      bit(Z_AXIS)
#define DEFAULT_HOMING_CYCLE_1      bit(Y_AXIS)
#define DEFAULT_HOMING_CYCLE_2      bit(X_AXIS)

thanks ^^

qiwalker, 2020-12-19:
hi,

thankyou for ur reply,

that's good! ^^
yes main coreXY still on XY
i'll be wait, hopefully extra axis can be add in corexy machine

anyway i can't use homing on coreXY
already add those script on my machine config and enable homing in machine settings but still not working

#define USE_CUSTOM_HOMING
#define DEFAULT_HOMING_CYCLE_0 bit(Z_AXIS)
#define DEFAULT_HOMING_CYCLE_1 bit(Y_AXIS)
#define DEFAULT_HOMING_CYCLE_2 bit(X_AXIS)

thanks ^^

bdring, 2020-12-19:
This was added today on a test branch. We also found some reporting issues. We want to make a few more changes and then will push to the active branches.

Please wait until early next week.

qiwalker, 2020-12-19:
thank you so much @bdring 
keep health.....^^,

bdring, 2020-12-19:
If you want to try an alpha version look at the AntFix branch. It supports CoreXY + ZABC (base on you N_AXIS setting. 

I am going to close this issue.

closed 2020-12-19 by bdring