bdring/Grbl_Esp32 #712: DC Encoder Motors - No Hardware Discussion!

InvncibiltyCloak, 2020-12-18:
Hey all, 
I am working on implementing DC encoder motor drivers directly. I am creating a new issue separate from the previous ones because it derailed into discussion on different motor/encoder options. No hardware discussion is allowed here!

I am from the MaslowCNC project and have gone ahead and made a board with 5 dc motor drivers. If anyone is interested in the details, the design can be found at [InvncibiltyCloak/ESPMaslowController](https://github.com/InvncibiltyCloak/ESPMaslowController). There are different kinematics on this machine, but that is simple to implement in the relevant machine file, so we can ignore this for now. 

The TLDR here is I have 5 encoders connected directly to the ESP32 and 5 DC motor drivers connected through an SPI-connected PWM generator. At this time, I am only trying for 3 axis control. 

The ESP32 has a Pulse Counter peripheral which is capable of doing quadrature decoding and counting to 16-bits. With some interrupts it can be extended to any size at minimal CPU load. This is how the encoders are counted.

I have been looking through the source for how best to handle closing the loop on the DC motors. It looks like stepper_pulse_func is where all of the action happens. 

I plan to use Motor.step() to allow my custom DCMotor class to count the generated step/direction pulses and PID based off that. Is there a more direct way to get the total step count?

Next, the PID requires a consistent timing for good performance. The main stepper_pulse_func gets called at varying timing depending on various factors, and in general is called far too often. At this point I plan to use TimerG0, Timer 1 with a different alarm value to drive the PID updates.

Any comments or suggestions on the plan? Thanks for the help!

InvncibiltyCloak, 2020-12-18:
And to answer the question "Is there a more direct way to get the total step count?" the answer was given by @BarbourSmith in one of the old threads - `sys_position`. So the plan has changed to use that instead.

BarbourSmith, 2020-12-19:
I am excited to hear that!

bdring, 2020-12-19:
In the heart of the firmware, it works only in steps in machine space. All units, offsets, etc. are applied as needed to inputs and outputs.

`int32_t sys_position[6]` is the array holding the current position.

bdring, 2020-12-19:
Here is an invite to our discord Server

https://discord.gg/zex6DqHa (expires in 1 day)