bdring/Grbl_Esp32 #715: JOG A Axis using  AntFix branch COREXY machine

qiwalker, 2020-12-21:
hi, @bdring 

i already use firmware AntFix
for XYZ axis working normal move using G91
when I moving A Axis using G91 it moving extremely slow
 
when using G90 XYZA axis working normal

thanks

bdring, 2020-12-21:
I seem to get he same speeds regardless of G90/G91 mode.

Is $MaxRate the same for all axes?

Send ...
```
G90
G0
X0
X300
G91
X-300
```

Does it move at different rates during the (2) 300 moves?

qiwalker, 2020-12-22:
$MaxRate is same for All Axis

using this code :

G90
G0
A0
A300
G91
A-300

Working Fine...
just when try to use G1 and set feedrate the A Axis move very slow even use maximum feedrate value

qiwalker, 2020-12-22:
Another Report Trial using core XY

send 
G0A1000

when A axis still moving send Ctrl+x (soft reset)
and the send $X

send again G0A1000
error found : when A Axis Moving, Motor 1 and Motor 2 for XY Axis also moving
Motor 2 also make noise (ringing)

bdring, 2020-12-22:
I can confirm the second problem you are reporting.  

If XY are at -1.000, -1.000 in CoreXY, the motors are actually at -2.000, 0.000. After the reset, the XY are targeting -2.000, 0.000 but applying CoreXY inverse kinematics to that. 

I have to see what reset is doing and fix that.

qiwalker, 2020-12-23:
ok i'll be wait for your next update and do another trial for CoreXY machine.
thankyou and keep health  ^^

qiwalker, 2021-01-06:
hi...

is there any update ?

thanks ^^

bdring, 2021-01-06:
I have not had time to work on this. It is a complicated issue. If you reset, you should home to restore the position for now. I hope to look at in around the middle of the month.

qiwalker, 2021-01-06:
ok, thanks for your prompt reply ^^
and also kindly check problem using G1 for ABC axis

thanks & regards