bdring/Grbl_Esp32 #723: Question: Is there a board that works as drop-in replacement for 3018-style desktop CNCs?

sgofferj, 2020-12-28:
I found this project while googling for CNC upgrades and I'm fairly enthusiasic about it! Awesome work @bdring !
However, it seems the only readily available boards are the ones on Tindie from Bart himself and they are maybe a bit "oversized" for a 3018-style desktop CNC plus none seems to have a simple spindle driver on board. I found the Woodpecker32 design but nobody seems to sell ready made Woodpecker32 boards.

Does anybody know of other options? Although I'm quite well versed with a soldering iron, I'd prefer a ready made board because SMD soldering isn't my favorite task...

joedirium, 2020-12-28:
What do you mean by oversized? Physical dimensions or price?
The 6-pack board with all of it's options appeals pretty universal to me.

sgofferj, 2020-12-28:
I mean feature-wise. That 6-pack board feels totally overkill for a small 3018-type - and as far as I understand, there's no DC spindle driver module available for that.

MitchBradley, 2020-12-28:
The 6-pack design was the answer to the continuing problem of having to make an endless array of special-purpose boards for "this machine with that special feature".  With 6-pack, the answer to "can I do X" - where everybody wants to do a different "X", is generally yes - although in some cases it might require a new module for a particularly odd wrinkle.  Perhaps someone will design exactly the board you want with no extra frills, and make a few units and you can get your hands on one.  Or you can design your own and have it fabbed by, say, JLCPCB or PCBWay including assembly of the SMT components.

joedirium, 2020-12-29:
I think Mitch is right.
If price and size is not the problem, why not just go for the 6-pack board. You might reuse it for future projects if you ever plan to replace your current CNC.
I do not know your machine, but I am pretty sure you find the right spindle driver for your spindle which can then be controlled by 0-10 voltage out module - if it support RPM control at all and does not simply need to pe powered.
You can print a case for everything if you own a 3D printer.
Also you will have to do some wiring.
After all you will have to spend quite some time and effort to use a different controller for you CNC. 

So right, there is no perfect matching plug and play solution for your CNC.

bdring, 2020-12-30:
What do you think about a controller like this?

[image]

It has one socket for a CNC I/O Module. These are the same ones used on a 6 Pack. This would allow the use of just about any type of spindle. The 5V module (shown) could support both a PWM spindle and a laser.

Everything else is built in.

Sell price would be about $60.

sgofferj, 2020-12-30:
I'd probably even happily buy the "six pack" controller if there was a module for a 24V DC spindle... I don't think, I'm yet ready for bigger modifications. Replacing the board is one thing but starting to rewire the spindle, let alone install another one, is probably above my current  skill levels. I'm just getting started with CNCing.

bdring, 2020-12-30:
Pololu has a lot of brushed DC motor drivers.

https://www.pololu.com/category/11/brushed-dc-motor-drivers

This looks like it might be right for a typical 24VDC spindle

https://www.pololu.com/product/1451

[image]

That could probably made into a module with some nice connectors.

[image]

If someone wants to sponsor a project like this, we could probably get some made.

rlwoodjr, 2021-02-09:
I m curious, how would I find out more about sponsoring a project?

bdring, 2021-02-09:
The design of the motor module is done. I have not had the time to order it. Too many other things going on.

[image]

[image]

whoim2, 2021-06-11:
im create "lite" board version and controller for 110v 800w D52mm 20k rpm DC spindel from aliexpress, link in video description. Controller use hall or optical rpm sensor for arduino or 3d-printer.
Now im order pcb and wait shipping.

First test spindel pcb: https://youtu.be/4QquvDP9aSU and  https://youtu.be/VMW0-LLrZDo

Im put files to github after fully test it.
Boards:

https://i.postimg.cc/pX5v9xFh/Screenshot-20210611-030852.png