bdring/Grbl_Esp32 #724: SD card "Refresh" without SD card inserted into MidTBot prevents further commands from executing with "Error 120"

Leemonoid, 2020-12-28:
What version of the firmware are you using?
1.3a 20201212

Is the problem repeatable?
Yes

Under what conditions does the bug occur?
Without an SD card inserted into the MidTBot, click the "Refresh" button in the Web GUI.  Eventually a "No SD card inserted" message will appear in the SD card window, but attempting to execute any further commands (such as $X) will report "Error 120 [MSG:SD card job running]."  The only way I've found to get past the error is to reset the ESP32 manually.

bdring, 2020-12-30:
I have confirmed the problem and have added it to our issue tracker. It is probably a simple fix that can be included in the next update.

AlphaCircuit, 2021-02-10:
Solutions as below.. 
I have made a solution to this problem. As follows modify as follows sdcard.cpp locate "get_sd_state" and  replace with below, I was struggling for days and realised all the prechecks on card being there where causing it not bother checking for a card after a no card refresh... this resolves it by taking out any notion of not checking ie it always checks ... note copying as vebatim as a lot is remarked out but left incase someone finds a better way.  Note it WORKS as is copy and paste it .... Cheers 

Modify File SDCard.CPP 

uint8_t get_sd_state(bool refresh) {
 /*
	if (SDCARD_DET_PIN != UNDEFINED_PIN) {
        if (digitalRead(SDCARD_DET_PIN) != SDCARD_DET_VAL) {
            sd_state = SDCARD_NOT_PRESENT;
            return sd_state;
            //no need to go further if SD detect is not correct
        }
    }

  if (!refresh) {
	  return sd_state;  //to avoid refresh=true + busy to reset SD and waste time
  }
  /////// Norm mod
    //if busy doing something return state
/*
  if (!((sd_state == SDCARD_NOT_PRESENT) || (sd_state == SDCARD_IDLE))) {
         return sd_state;
    }

  if ((sd_state == SDCARD_NOT_PRESENT)) { //|| (sd_state == SDCARD_IDLE))) {
	  SD.end();
	  return sd_state;
  }

   if (!(sd_state == SDCARD_IDLE)) {
      return sd_state;
	}
		
	
	/////////////Norm Mod ends
*/
    if (!refresh) {
        return sd_state;  //to avoid refresh=true + busy to reset SD and waste time
    }

    //SD is idle or not detected, let see if still the case
   // SD.end();
   // sd_state = SDCARD_NOT_PRESENT;

    //using default value for speed ? should be parameter
    //refresh content if card was removed
    if (SD.begin((GRBL_SPI_SS == -1) ? SS : GRBL_SPI_SS, SPI, GRBL_SPI_FREQ, "/sd", 2)) {
	    if (SD.cardSize() > 0) {
            sd_state = SDCARD_IDLE;
        }
    }
    return sd_state;
}

bdring, 2021-02-10:
We have made some SD card changes recently. Have you looked at the latest code?

bdring, 2021-02-10:
The best way is to create a pull request. It makes it very easy for us to see the impact on the current code. We have guidelines here.

https://github.com/bdring/Grbl_Esp32/wiki/Pull-Request-Guidelines

Leemonoid, 2021-02-13:
Closing this issue.  With the latest firmware build, the problem has been corrected.  Thanks!

closed 2021-02-13 by Leemonoid