bdring/Grbl_Esp32 #725: Problem with 6 axis servo machine

AndreasEpp, 2020-12-30:
Hello,
I can't get my 6 axis servo machine (all 6 axis are servos) running. The servos don't move no matter which g-code I send.
The Hardware is okay I think, because all 6 servos move with a simple Arduino sketch with ledcWrite and so on.
The controller itself is also okay because I can access it via serial, telnet and the webpage.
My machine file looks as follows and is included in the Machine.h file. 

```
#define MACHINE_NAME            "6AxisRobot"

#define N_AXIS 6

#define SERVO_X_PIN    			GPIO_NUM_13
#define DEFAULT_X_MAX_TRAVEL    120.0
#define DEFAULT_X_HOMING_MPOS   60.0
#define X_SERVO_CAL_MIN         1.0
#define X_SERVO_CAL_MAX         1.0

#define SERVO_Y_PIN    			GPIO_NUM_27
#define DEFAULT_Y_MAX_TRAVEL    120.0
#define DEFAULT_Y_HOMING_MPOS   60.0
#define Y_SERVO_CAL_MIN         1.0
#define Y_SERVO_CAL_MAX         1.0

#define SERVO_Z_PIN             GPIO_NUM_26
#define DEFAULT_Z_MAX_TRAVEL    120.0
#define DEFAULT_Z_HOMING_MPOS   60.0
#define Z_SERVO_CAL_MIN         1.0
#define Z_SERVO_CAL_MAX         1.0

#define SERVO_A_PIN    			GPIO_NUM_25
#define DEFAULT_A_MAX_TRAVEL    120.0
#define DEFAULT_A_HOMING_MPOS   60.0
#define A_SERVO_CAL_MIN         1.0
#define A_SERVO_CAL_MAX         1.0

#define SERVO_B_PIN             GPIO_NUM_33
#define DEFAULT_B_MAX_TRAVEL    120.0
#define DEFAULT_B_HOMING_MPOS   60.0
#define B_SERVO_CAL_MIN         1.0
#define B_SERVO_CAL_MAX         1.0

#define SERVO_C_PIN             GPIO_NUM_32
#define DEFAULT_C_MAX_TRAVEL    120.0
#define DEFAULT_C_HOMING_MPOS   60.0
#define C_SERVO_CAL_MIN         1.0
#define C_SERVO_CAL_MAX         1.0
```

My boot message 

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Grbl_ESP32 Ver 1.3a Date 20201124]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:6AxisRobot]
[MSG:Axis count 6]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:No spindle]

[MSG:Client Started]
[MSG:Connecting MYROUTER]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connecting.]
[MSG:Connected with 192.168.178.74]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]

Grbl 1.3a ['$' for help]

```

As for my use case:
I am trying to get one of these cheap chinese 6 dof robot arms to work with grbl_esp32.

Thanks in advance!

bdring, 2020-12-30:
Change like this on all servos.  #define SERVO_X_PIN  to #define X_SERVO_PIN 
 
It should, work, but this has never been tested before. It was not optimized to do this. Each servo uses an RTOS task that takes up  memory. 

If you have crashes we could rewrite it to only use one task for any amount of servos.

AndreasEpp, 2020-12-30:
Thank you for your fast reply!
They are moving now :)

I've got the wrong define from the template file [here](https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Machines/template.h#L134).
Maybe this needs some fixing ;)

A small problem remains though. It looks like the limits are not set correctly:

```
[MSG:Grbl_ESP32 Ver 1.3a Date 20201124]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:6AxisRobot]
[MSG:Axis count 6]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:X  Axis RC Servo Pin:13 Pulse Len(3276,6553) Limits(60.000,180.000)]
[MSG:Y  Axis RC Servo Pin:27 Pulse Len(3276,6553) Limits(60.000,180.000)]
[MSG:Z  Axis RC Servo Pin:26 Pulse Len(3276,6553) Limits(-60.000,60.000)]
[MSG:A  Axis RC Servo Pin:25 Pulse Len(3276,6553) Limits(-60.000,60.000)]
[MSG:B  Axis RC Servo Pin:33 Pulse Len(3276,6553) Limits(-60.000,60.000)]
[MSG:C  Axis RC Servo Pin:32 Pulse Len(3276,6553) Limits(-60.000,60.000)]
```

Is it possible to set them all to 0...120 or -60...60 ?
60...180 is odd to handle.

Otherwise it works like a charm! The Connection over Telnet is stable and no crashes so far.

bdring, 2020-12-30:
Here is a good reference on machine range.

https://github.com/bdring/Grbl_Esp32/wiki/Machine-Space-and-Homing

AndreasEpp, 2021-01-01:
Hi,
that did the trick, thank you!

For anyone searching:
$23=0
will change the homing direction and thus change where the axis "starts"
=> all axis report from -60 to 60 now

Thank you for your help and your work on grbl_esp32!

closed 2021-01-01 by AndreasEpp