bdring/Grbl_Esp32 #728: Limit Invert Mask

reynolds087, 2021-01-03:
Hi, I see this line in the roadmap under recently completed:

Automatically generate #define LIMIT_MASK from defined pins.

Does that mean that the feature is available, or is it still in progress? I have a mixture of normally open and normally closed limit switches. It would be difficult to change that. If not, is there another work around for my situation?

bdring, 2021-01-03:
See [this wiki page](https://github.com/bdring/Grbl_Esp32/wiki/Help-with-Switch-Problems#inverted-states) on how to do that.

**Stay tuned:** This is about to get way better in the next major update.

reynolds087, 2021-01-03:
Thanks, is the example on that page meant to be literal? for instance if I want to only invert the x and z limit switches, would I use this line?

`#define INVERT_LIMIT_PIN_MASK ((1<<X_LIMIT_BIT)|(1<<Z_LIMIT_BIT))`

reynolds087, 2021-01-03:
It won't compile the way it's shown on that page. I'm sure I'm doing something wrong, just can't figure out what. I copy pasted the example line from the wiki and get this message:

sketch\src\Limits.cpp: In function 'AxisMask limits_get_state()':

sketch\src\Machines/josh.h:108:36: error: 'X_LIMIT_BIT' was not declared in this scope

 #define INVERT_LIMIT_PIN_MASK ((1<<X_LIMIT_BIT)|(1<<Y_LIMIT_BIT))

                                    ^

sketch\src\Limits.cpp:331:16: note: in expansion of macro 'INVERT_LIMIT_PIN_MASK'

     pinMask ^= INVERT_LIMIT_PIN_MASK;

                ^

sketch\src\Machines/josh.h:108:53: error: 'Y_LIMIT_BIT' was not declared in this scope

 #define INVERT_LIMIT_PIN_MASK ((1<<X_LIMIT_BIT)|(1<<Y_LIMIT_BIT))

                                                     ^

sketch\src\Limits.cpp:331:16: note: in expansion of macro 'INVERT_LIMIT_PIN_MASK'

     pinMask ^= INVERT_LIMIT_PIN_MASK;

reynolds087, 2021-01-03:
I guess I was overthinking it. Set it like other axis masks, and it worked. I put 2, because I actually only needed to invert my Y axis.

closed 2021-01-03 by bdring