bdring/Grbl_Esp32 #731: Stepper motors always locked when step idle delay changed.

MillerRen, 2021-01-05:
When i change the 'step idle delay' to 255. stepper motors locked. It's ok. But when i change it back to 25. stepper motors still locked. The step singal allways high.

My firmware version is 1.3a

[image]

```C++
 #define X_STEP_PIN              GPIO_NUM_14
#define X_DIRECTION_PIN         GPIO_NUM_27
#define Y_STEP_PIN              GPIO_NUM_26
#define Y_DIRECTION_PIN         GPIO_NUM_25
#define Z_STEP_PIN              GPIO_NUM_33
#define Z_DIRECTION_PIN         GPIO_NUM_32

#define X_LIMIT_PIN             GPIO_NUM_16
#define Y_LIMIT_PIN             GPIO_NUM_4
#define Z_LIMIT_PIN             GPIO_NUM_15

#define DEFAULT_HOMING_CYCLE_0 bit(Z_AXIS)
#define DEFAULT_HOMING_CYCLE_1 bit(X_AXIS)

// OK to comment out to use pin for other features
#define STEPPERS_DISABLE_PIN    GPIO_NUM_12

// Spindle Output Pin
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_17

#define SPINDLE_TYPE     SpindleType::LASER 
#define USE_MACHINE_INIT

```

bdring, 2021-01-05:
I think they will unlock after the next move.

If you want to unlock them temporarily use $MD

https://github.com/bdring/Grbl_Esp32/wiki/Settings#motordisable-or-md

MillerRen, 2021-01-05:
Thanks for your reply. I have tried many times but problem still even though erase full flash and reflash esp32. Maybe my board causing. I will check it. Thx.

bdring, 2021-01-05:
I would assume you have a hardware or machine definition problem. Nobody has reported an issues with stepper enable. 

What hardware are you using. Are you 100% sure the pin is connected properly?

Try this.

Set $stepper/IdleTime=200
Reboot the ESP32
Check the voltage of GPIO_NUM_12
Do a long move while checking the voltage of 12

sjonholle, 2021-01-08:
GPIO12 is a special pin https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection
Have you tried GPIO13 ? the examples use also this pin.

MillerRen, 2021-02-06:
resolved by new code.

closed 2021-02-06 by MillerRen