bdring/Grbl_Esp32 #735: GPIO_NUM_2 as an input for cycle start/resume

JHarding86, 2021-01-11:
Hello amazing Grbl_Esp32 gurus! Thanks for the port of Grbl to a more capable system, it is refreshing.

I have been using Grbl with a 328p for a while now and figured it was time for an upgrade. I took my Prontoneer v3.0 shield off of my Arduino Uno and stuck it onto an ESPDuino 32. This worked surprisingly well. X,Y, & Z steppers and end stops work out of the box without issue! yay!

Now, I am am trying to get the cycle start/resume functionality working. I have a CONTROL_SAFETY_DOOR_PIN assigned, and this properly pauses the machine stopping all motion and turning off my laser. I assigned CONTROL_CYCLE_START_PIN to GPIO_NUM_2, thinking that this would work as an input for my resume button. This is not really the case, it seems finicky at best. Sometimes triggering when expected, sometimes triggering seemingly on its own. My understanding is GPIO_NUM_2 is a touch sensor? will this work with a push button switch as well?

I have attached my machine configuration file as well as my config.h file. Please bear with me as I am new to the ESP32 world...

[Config.txt](https://github.com/bdring/Grbl_Esp32/files/5798488/Config.txt)
[wemos_d1_r32_esp_wroom_32.txt](https://github.com/bdring/Grbl_Esp32/files/5798489/wemos_d1_r32_esp_wroom_32.txt)

MitchBradley, 2021-01-11:
Read what this page https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins says about GPIO 2

JHarding86, 2021-01-11:
Ahhh. The page we have all been looking for! Thanks for the pointer.

One more question, can you confirm that the ESP32 uses 3.3v logic? I have had to switch my trigger from a 5v to 3.3v source for pin 35 to trigger my input pin. This seems strange to me because all of my limit switches work just fine???

bdring, 2021-01-11:
The ESP32 is a 3.3V part. Use of 3.3V on I/O could damage it.

JHarding86, 2021-01-11:
I assume you mean use of 5v on I/O could damage it...

closed 2021-01-12 by JHarding86