bdring/Grbl_Esp32 #736: Pin D11 unusable as step pin on ESP32D0WDQ6  with RMT

shorai, 2021-01-12:
**Description**

- When using pin D11 as a step pulse pin the firmware fails on startup with a repeated reboot.
- When the step function is reassigned to another pin, the firmware works perfectly.
- Pin D11 can be used for other functions, in which case the firmware works correctly.
- The unavailability of Pin 11 for RMT may be an undocumented feature of ESP32 

I found some reference to someone unable to use RMT with more than one channel.(not the case) 
Perhaps they unknowingly stumbled into this or a similar problem and were unable to resolve it.

**Priority**
Low, an immediate workaround exists - avoid D11 as a step pulse pin.

**What version of the firmware are you using?**
```
[MSG:Grbl_ESP32 Ver 1.3a Date 20201212]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:Rockwell]
[MSG:Axis count 4]
[MSG:RMT Steps]
```
The modules are from Banggood ESP-wroom-32
Under Arduino the ESP modules  identify as  
```
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
```

**Is the problem repeatable?**
- The problem is repeatable and tested on two different ESP modules
- It can be tested on a bare board without connections to a physical machine
- It is repeatable under Arduino 1.8.13 
                                               `   arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/`
                            and PIO latest release downloaded 11 January 2021
                                                 `[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]`
- I had the same problem with 3 axis configuration

**Under what conditions does the bug occur?**
Startup

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Grbl_ESP32 Ver 1.3a Date 20201212]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:Rockwell]
[MSG:Axis count 4]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:X  Axis Standard Stepper Step:GPIO(13) Dir:GPIO(12) Disable:GPIO(21) Limits(-500.000,0.000)]
Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment). Exception was unhandled.
Core 1 register dump:
PC      : 0x4018d1de  PS      : 0x00060630  A0      : 0x800d5b38  A1      : 0x3ffd30b0  
A2      : 0x00000002  A3      : 0x3ffc5cac  A4      : 0x00000002  A5      : 0x00000000  
A6      : 0x3ff56000  A7      : 0x00000000  A8      : 0x00000006  A9      : 0x3ff56000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000002  A13     : 0x00000000  
A14     : 0x00060420  A15     : 0x00000000  SAR     : 0x00000015  EXCCAUSE: 0x00000009  
[... 309 lines cut ...]
#define DEFAULT_HOMING_SEEK_RATE 120.0  // $25 mm/min
#define DEFAULT_HOMING_PULLOFF 5.0  // $27 mm

// set motrors to 12800 steps /rev
#define DEFAULT_X_STEPS_PER_MM 4266.667 
#define DEFAULT_Y_STEPS_PER_MM 4266.667
#define DEFAULT_Z_STEPS_PER_MM 4266.667
#define DEFAULT_Z2_STEPS_PER_MM 4266.667

#define DEFAULT_X_MAX_RATE 1500.0  // mm/min
#define DEFAULT_Y_MAX_RATE 1500.0  // mm/min
#define DEFAULT_Z_MAX_RATE 1200.0  // mm/min

// ============== Axis Acceleration =========
#define DEFAULT_X_ACCELERATION 200.0
#define DEFAULT_Y_ACCELERATION 200.0
#define DEFAULT_Z_ACCELERATION 200.0

// ========= AXIS MAX TRAVEL ============

#define DEFAULT_X_MAX_TRAVEL 500.0  // $130 mm NOTE: Must be a positive value.
#define DEFAULT_Y_MAX_TRAVEL 300.0  // mm NOTE: Must be a positive value.
#define DEFAULT_Z_MAX_TRAVEL 180.0  // mm NOTE: Must be a positive value.

```

rafl, 2021-01-12:
Note how https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins suggest to not use GPIO_NUM_11 and a number of other GPIOs because of their use for other purposes.

MitchBradley, 2021-01-12:
Closing since this is a known restriction of the hardware. Thanks for providing the detailed information.

shorai, 2021-01-12:
Thanks for the quick response.

Apologies for having missed that document.

closed 2021-01-14 by bdring