bdring/Grbl_Esp32 #737: question about machine.h files

sixtwoeight, 2021-01-15:
does the 6 axis v1p5 have the same base physical pinout for as the 3 axis v4.1… in other words will machine.h profile for the 3 axis work on the 6 axis (with the exception of the remaining axis)?

I was a bit overzealous when choosing the board i need, and got the 6 axis when at the moment all i really need is the 3 axis.  My understanding is that the 3 axis doesn’t need so much pin mapping out of the box. I do have future plans that will utilize the additional 3 axis however, until my esp32 skills level grows a bit more all i really needed was the 3 axis.  my hope is that i can just use the machine definition file for the 3 axis on the 6 axis everything automagically work. is that possible?

thanx in advance and thank you for all the brilliant work ;)

bdring, 2021-01-15:
There are several examples of 6 pack machine definitions in the [Machines](https://github.com/bdring/Grbl_Esp32/tree/main/Grbl_Esp32/src/Machines) folder. Here is a [3 axis one](https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Machines/6_pack_stepstick_XYZ_v1.h).

The [6 Pack Wiki](https://github.com/bdring/6-Pack_CNC_Controller/wiki/Socket-Pin-Number-Mapping) has that complete pin mapping of the 6 pack.

sixtwoeight, 2021-01-17:
excellent thank you, that looks like it will work perfectly.  and since i also got the 4x switch input, and am using optical endstops so i just need to add "

#define DEFAULT_INVERT_LIMIT_PINS       1

under

#define Z_LIMIT_PIN                 GPIO_NUM_35

and place the module in socket #1?

thank you again for a great product.

clinton james russell

bdring, 2021-01-18:
Correct

closed 2021-01-18 by bdring