bdring/Grbl_Esp32 #738: Door pin

PinoS56, 2021-01-16:
Hello,
best job grbl_esp32, 
I would have a problem with this:
The pin of the door works in positive logic with the switch normally open, it would be better to use it in negative logic with the switch normally closed, so that if it breaks it would be in alarm, there is the possibility to invert the state of the pin. .?
sorry my bad inglish
 @thanks...

bdring, 2021-01-16:
The door pin logic can be inverted in the config.h file.

https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Config.h#L57

The next major release of Grbl_ESP32 will enable all pins like this to be inverted without a recompile.

PinoS56, 2021-01-16:
Solved thanks...

closed 2021-01-16 by PinoS56