bdring/Grbl_Esp32 #739: Ganged Axes & Axis Squaring switch set up.

rlwoodjr, 2021-01-16:
I am interested in setting up a gantry machine to home each side independently.  I understand how to set up the firmware, but I am unclear on the home switch connections.  Our gantry axis is the X axis.

Does this work the same as GRBL1.1 where one of the X home switches share the same IO pin as the Z axis?  

If not, with one switch how, does the firmware determine which X motor to stop when a single switch is activated?

bdring, 2021-01-16:
It is rather simple. Both switches are on the same circuit. It homes both motors until the circuit is triggered. It then homes each motor by itself until the circuit is triggered. It just stops the moving motor(s).

rlwoodjr, 2021-01-16:
Awesome. Thanks for the clarification. I am looking forward to trying this on our KL7 series CNCs.

closed 2021-01-16 by bdring