bdring/Grbl_Esp32 #743: 3 axis with rotating axis

rlwoodjr, 2021-01-19:
I am working on a machine header file that will allow for a XZA CNC.  Only 3 axes, but one of them rotating.  It loads to the ESP32, but only shows 2 axes  X and Z.

If I define a 4 axes CNC XYZA, then it works as expected.  What am I missing?

This is the XZA.h file

#define MACHINE_NAME            "XZA"

#define N_AXIS 3

#define X_STEP_PIN                   GPIO_NUM_12
#define X_DIRECTION_PIN         GPIO_NUM_14
#define A_STEP_PIN                   GPIO_NUM_26
#define A_DIRECTION_PIN         GPIO_NUM_15
#define Z_STEP_PIN                   GPIO_NUM_27
#define Z_DIRECTION_PIN         GPIO_NUM_33

#define X_LIMIT_PIN             GPIO_NUM_17
#define A_LIMIT_PIN             GPIO_NUM_4
#define Z_LIMIT_PIN             GPIO_NUM_34

bdring, 2021-01-19:
You cannot skip an axis. If you want XZ and A you must have

`#define N_AXIS 4`

You do not have to define anything for the Y axis. It will create a virtual axis for that. You can pretend it does not exist.

rlwoodjr, 2021-01-19:
Thanks. I will give that a try.

closed 2021-02-09 by rlwoodjr