bdring/Grbl_Esp32 #745: firmware confusion

sixtwoeight, 2021-01-19:
i don't think this is a compiling error because it seems that i haven't gotten around to that.. 

I feel a little confused about compiling the firmware. i have followed the instructions for setting up the arguing ide for the esp and that seems to work fine (worked fine for flashing the firmware for an esp cam).  and i believe i got the machine.h profile set up correctly with my current machine definition. but when i load the sketch included with the GitHub files it seems pretty blank.  yet when i watch a YouTube video of someone flashing a controller for grbl_esp when they open the sketch or the first time it comes preloaded with #includes and all (multiple tabs open etc) but when i open it it’s pretty much empty.  it looks similar to the picture of what settings should look like for the esp in GitHub.  but when i open the sketch it’s just empty.  what am i missing?

sorry if this is very basic, i'm just starting out.

richardwpaul, 2021-01-19:
I had the same issue and ended up editing in Notepad++ while using arduino IDE to compile

Rick Paul
mrp email

bdring, 2021-01-19:
The firmware is standard C++. We have an almost empty Grbl_ESP32.ino file to make it work with the Arduino system.

You should [follow the wiki](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware) on how to edit the [Machine.h](https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Machine.h) file.

You should be able to compile the firmware from the Arduino IDE. The developers use Platform.io in Visual Studio Code. It is a better IDE, but we do not provide support on how to set that up.

sixtwoeight, 2021-01-19:
I think i got the machine.h correct. if i understand it correctly all i had to do was add my machine definition file. or am i not getting some part of the instructions.  i've read the wiki over and over, if i'm missing something it's probably because i'm new to programing in general-i have a tendency of biting off more than i can chew.

where i'm i'm lost is in the arduino sketch, do i not add anything do that? i just modify the machine definition file. then make sure that's included in the machine.h file and that's it? i don't need to modify the Grbl_Esp32.ino sketch at all? i feel like i should be have all the files in the src directly added to the arduino sketch independently or something... i feel like being such a noob is making me miss something extremely basic that's obvious to everyone else.

bdring, 2021-01-20:
Yes, you just change the machine file and upload. You could upload the default (no change) if you want to test drive it with no motors, etc.

sixtwoeight, 2021-01-20:
got it.. worked.. thank you again so much... for ur help and awesome controller ;)

sixtwoeight, 2021-01-25:
so i for sure bit off more than i could chew.  i missed your email from when i purchased the board, which was a big mistake.  i got it to flash, then the motors didn't work, then i realised it's because i'm using tmc2209's.. so all morning i've been comparing machine definition files and trying different things to no avail.. so i was hoping its not to late to reply to that mail and if you might be able to help put it together i could flash it.   i'm just way to much of a noob for this.. rofl.  anyway it would be awesome if you could..
anyway here's the response to that email;

  *   Motors
     *   How many lettered axes (not motor count) do you plan to use?​ 3
     *   Will any axes use 2 motors? ​NO
     *   What type of motor drivers are you using? TMC2209
  *   Switches
     *   How many limit switches will you be using? 3
     *   Are you using a Z Probe? YES
     *   Anything else? NO
  *   Spindle
     *   What Type of Spindle/Laser do you want A CHEAP 500 “QUIET CUT SPINDLE” WITH A POTENTIAL, IN THE POWER SUPPLY
  *   Accessories
     *   Are there any accessories you want to control? (Vacuum, Coolant, etc) NOT AT THE MOMENT
  *   Unused Modules
     *   Did you purchase any modules that you do not plan to use right away? YES, THE 4 INPUT I/O MODULE

Is there anything I left out? Did you purchase any module for special use not covered above. NOPE THAT COVERS IT

thank you again so very much...
clinton...

sixtwoeight, 2021-01-26:
i just realised that i should have been more specific about the limit switches i'm using optically normally closed

bdring, 2021-01-26:
Do you have an order number I can use to verify the parts you have?

The TMC2209 is only supported in stand alone mode. Do you have a link to where you bought them, so I can verify.

Not all optical vane switches work well with the 4x input module. It has optical isolation and some of the basic switch circuits designed for 3d printers have trouble. You can try, but it have have trouble. Regardless the firmware would be the same. The N.C. vs. N.O. can be change without recompiling.

sixtwoeight, 2021-01-26:
sure thing..

Order #238312

here is a link along with the data sheet:

https://www.digikey.com/en/products/detail/trinamic-motion-control-gmbh/TMC2209-SILENTSTEPSTICK/10232488

and i also have mechanical switches. if they word better then not a problem... it would still be normally closed....

thank you in advance
clinton

bdring, 2021-01-26:
I will make you a machine definition file later today.

There is a wiki page regarding TMC2209 drivers.

https://github.com/bdring/6-Pack_CNC_Controller/wiki/TMC2209-Stepsticks

sixtwoeight, 2021-01-26:
lol, thank you so very much...  i also purchased 3 tmc5160's (before i even got ur controller... would using them be easier... i know that the board has had issues with them... the ones i got are also from digikey ( https://www.digikey.com/en/products/detail/trinamic-motion-control-gmbh/TMC5160-SILENTSTEPSTICK/9990281 )

would that be a better choice? i feel like it would... was just going to use the 2209's because i saw the posting about the 5160's

lol.... thank you so much for ur help ....

https://github.com/bdring/6-Pack_CNC_Controller/wiki/TMC2209-Stepsticks

bdring, 2021-01-27:
Here is your machine definition file. I have not tested TMC2209 in StepStick mode. I just used the data sheet for reference.

I would recommend TMC2130 or DRV8825 if you are looking for other drivers. You would need to change the machine file.

[6_pack_TMC2209_XYZP_SpinRelay.zip](https://github.com/bdring/Grbl_Esp32/files/5881498/6_pack_TMC2209_XYZP_SpinRelay.zip)

For future support, I suggest using Discord.

https://discord.gg/jKUMabHRKD

sixtwoeight, 2021-01-27:
just wanted to say thank you so much.... sorry for being a pain in the ass... thank you again... 🙂

https://discord.gg/jKUMabHRKD

closed 2021-01-27 by bdring