bdring/Grbl_Esp32 #746: ESP32 Keeps Resetting Early in Program

reynolds087, 2021-01-20:
It seems like my VFD might be causing the issue, but is there any obvious reason why the ESP32 would reset when it kicks on or after a short time running the spindle? Usually the program will only run for a couple of minutes before the ESP32 resets unpredictably. Sometimes it happens right when the VFD kicks on at the beginning of the program. Are there preventative measures that need to be taken to protect the ESP32? The VFD is on a separate circuit breaker, but I don't know if the magnetic interference could be propagating through the air or the wires in my house.

MitchBradley, 2021-01-20:
Magnetic interference does indeed propagate through the "air".  Actually it doesn't need air; magnetism propagates just fine through vacuum.  There are certain materials, typically dense metals, that will "short out" magnetic flux lines.  There are any number of techniques that can be used to reduce such interference, including ferrite beads, shielding of cables and enclosures, careful routing of cables to increase physical separation between interferers and interferees (and to put them at right angles), use of twisted pair signal cables (in properly connected twisted pairs, interfering signals cancel out between the two conductors).  This is a huge topic at an advanced level of electrical engineering.  Trial and error is usually necessary to solve it because the physical setup of every system is different.

reynolds087, 2021-01-20:
I took the VFD out of the case and ran it with them physically separated, and sure enough, it is working ok now. Would a faraday cage made out of some low cost wire mesh be a viable solution? I could connect an external antenna to the ESP32 for the WiFi features.

MitchBradley, 2021-01-20:
It might work, but I wouldn't give it a high probability.  Faraday cages are okay for reducing interference that is predominantly electrostatic in nature, but not great for magnetic isolation.

MitchBradley, 2021-01-21:
Physical separation is unlikely to fix a ground loop problem, unless the separation also involved a change in the ground wiring.

MitchBradley, 2021-01-21:
Since moving the VFD out of the case solves the problem, I would consider a mounting strategy where the VFD is away from the other electronics.  Magnetic fields fall off as the square of the distance, so physical separation can be very effective.

joedirium, 2021-01-21:
You might also use shielding in combination with the recommended proper grounding.
Some simple aluminum foil  connected to ground (minus) between VFD and board might help. If testing works you may print a cover for the VFD and glue the foil to it. A screw will then help to connect the ground wire.

I like VFDs. A bit retro, but still looking the best.

atrejo, 2021-03-05:
The ESP32 just loops restarting:

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9564
ho 0 tail 12 room 4
load:0x40080400,len:6320
entry 0x400806a8
ets Jun  8 2016 00:22:57

I did it with several boards, but esp32 get ID examples and others go fine, the problem is grbl_esp32 firmware

AllanCosta12, 2021-03-30:
> O ESP32 simplesmente reinicia em loop:
> 
> rst: 0x3 (SW_RESET), inicialização: 0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP: 0xee
> clk_drv: 0x00, q_drv: 0x00, d_drv: 0x00, cs0_drv: 0x00, hd_drv: 0x00, modo wp_drvIO: 0x00
> , clock div: 0x00 1
> carga: 0x3fff0018, len: 4
> load: 0x3fff001c, len: 1100
> load: 0x40078000, len: 9564
> ho 0 tail 12 room 4
> load: 0x40080400, len: 6320
> entry 0x400806a8
> ets Jun 8 2016 00:22:57
> 
> Fiz isso com várias placas, mas esp32 obtém exemplos de ID e outras funcionam bem, o problema é o firmware grbl_esp32

hello I would like to know if you managed to solve the error, because I am having the same problem, it will compia normal, but does not create the wifi network.
I don't know if the problem is the card, the firmware, or if I'm doing something wrong.

atrejo, 2021-03-30:
The problem is grbl_esp32 even running in test mode, it just keeps restarting

AllanCosta12, 2021-03-30:
I understand, but what did you manage to solve?
Well, I enter the file "machine.h" and changed it not to use in mode test_drive, commented the line that included "test_drive" and uncommented the line that included "3axis_v4.h", but the error insists.
I will do some sketches to see if any of the symptoms persist